opentx/opentx #3239: Feature request: Lua-configurable timeout on cell data

hymanroth, 2016-01-18:
Version 2.1

I notice that if the voltage sensor stops transmitting, the cell data on the radio persists the last values received. This makes sense to avoid blips, etc but it introduces another problem. Should the sensor fail, then the radio would be unaware and this could lead to battery alarms being missed.

So I propose a timeout parameter (which can set from Lua) after which the cell data is reset.

This would allow battery monitoring scripts to alert if cell telemetry is lost (even if general telemetry is still available)

projectkk2glider, 2016-01-18:
Internally each telemetry value has a validity bit that tells us if the data is fresh. It is currently not available to Lua (I think). We must think of a way to get that value to Lua.

lshems, 2016-01-18:
Not only lua I presume??

hymanroth, 2016-01-18:
@projectkk2glider Sounds good. Out of interest, what's the timeout on the validity bit?

projectkk2glider, 2016-01-18:
Timeout is 15 seconds.

hymanroth, 2016-01-18:
That sounds reasonable. I think it would be pretty straightforward to add and extra field to the table returned by Lua's getValue() for cells (https://opentx.gitbooks.io/opentx-lua-reference-guide/content/general/getValue.html) in which the stale bit is passed as a boolean.

closed 2020-07-07 by stale[bot]