opentx/opentx #3242: Improve telemetry layout

lshems, 2016-01-19:
In the current screen layout, all settings are in the top half screen, and all sensors ar in a big table below.

The settings for the 4 telemetry screens then "dangle" a bit lost below the sensor settings.

It would be cleaner to put them just above the sensor settings.

Just my opinion, and taste is debatable.

bsongis, 2016-01-20:
In the firmware (which radio)? Or in Companion?

lshems, 2016-01-20:
Companion.

het volgende:

kilrah, 2016-01-20:
On companion I agree that the sensor panel and telemetry screen panel could be swapped. 
Not on the radio though, as the sensor list allows for a quick glance to sensor values.

lshems, 2016-01-20:
Sounds great

fraternl, 2016-01-21:
Have you considered 2 tabs/screens for telemetry?

A telemetry input and a telemetry output page

elecpower, 2016-01-21:
Agree with @fraternl on the separation of purpose. The form is becoming a bit unwieldy.

projectkk2glider, 2016-01-25:
Should be done together with https://github.com/opentx/opentx/issues/2781#issuecomment-158743482

closed 2020-07-07 by stale[bot]