opentx/opentx #3244: Issue with overloading trim

fabiengagne, 2016-01-20:
A nice function to use trim for two or more different uses (overloading its function) is documented here.
http://rc-soar.com/opentx/setups/overloading_trims/index.htm
The method the author describes allows using the trim button to change the value of different GV (trim) while in flight. I find this very useful.

Unfortunately, there seems to be an issue when holding the trim button for a while (until it repeats) and then releasing. The last beep of the trim, when in overloaded function, is sent to the actual trim setting. There seems to be a transient effect.

I'm using OpenTx 2.1.7 on a Taranis.

fabiengagne, 2016-01-20:
PS. I can also reproduce the issue in the Companion 2.1 simulator.

Load the eepe file from the zip file here
http://rc-soar.com/opentx/setups/overloading_trims/index.htm
Open the SwitchTrim model.
Click on Simulate
Select the Gvars tab
SA down
Click on aileron-right-trim for 1.5 second and release
Observe the right trim will move toward the right by one position when releasing the trim button after that long press.
Otherwise, with short clicks on the trim buttons the function works well for changing any GV.

PPS There is also the same odd behaviour when single clicking rapidly on the trim buttons. Sometimes one of them is sent to the actual trim value instead of the GV.

projectkk2glider, 2016-05-02:
The problem is in the eepe you linked. The condition that turns on GVAR adjustment by TrmA is also dependent on the fact that the trim button is pressed. I don't know why Mike complicated his example so much (I will contact him to correct it).

The offending logical switches:
```
L1 AilTrim Left AND SA↓
L2 AilTrim Right AND SA↓

SF1 L1 Adjust GV1 TrmA ON 
SF2 L2 Adjust GV1 TrmA ON
```

The correct example would be (notice absence of logical switches):
```
SF1 SA↓ Adjust GV1 TrmA ON
```

RC-SOAR, 2016-05-02:
Agree this was way over the top! Web page is now updated with the corrected/simplified version.

http://rc-soar.com/opentx/setups/overloading_trims/index.htm

closed 2016-05-02 by projectkk2glider