opentx/opentx #3249: Changing voltage precision = wrong voltage trigger in logical switch

joeynuggetz, 2016-01-23:
Not sure if this is already reported. Running 2.1.7 on taranis x9d plus. If I change the precision of A2 from 0.0 to 0.00 in sensor setup it changes the voltage high and lows I set up in my logical switches. Decimal place is wrong. So example my voltage to trigger a sound at 10.4volts now reads 1.04. after changing voltage precision from 0.0 to 0.00.

kilrah, 2016-01-23:
That's normal, changing the precision changes how the value is interpreted, it doesn't change the value to adapt.
You typically set precision before doing anything else.

joeynuggetz, 2016-01-23:
Ok but it would make more sense if the value adapted. If i set a trigger on 10 volts why would i expect that a precision adjustment totally invalidate my value? It should improve it if anything.

kilrah, 2016-01-23:
Logical switch values are stored as integers, so say "123".
If precision is 0 that is interpreted as 123, if it's precision 1 it's interpreted as 12.3, precision 2 1.23.

The reason is mostly technical, but given it makes no sense offering a precision in the LS that is different from the actual value it's good as it is.

agibson2, 2016-09-15:
I just independently discovered this issue and ran into it on 2.1.x.  I made a change to precision (0.0 to 0.00) after I had the voltages configured in LOGICAL SWITCHES.  I luckily just happend to look at the LOGICAL SWITCHES area before flying and saw that my warning voltage was now 0.38 volts instead of 3.8v.  If I would have not looked at that I would have likely ran the battery low.  I always depend on voltage alerts for when to land.

closed 2019-11-15 by stale[bot]