opentx/opentx #3251: OpenTX Companion not flasing, but eepe flashing

GilDev, 2016-01-24:
Hi,
I have a Turnigy 9X, and I had Er9x installed on it for a while now, but I wanted to install OpenTX. Only problem, OpenTX Companion does not want to flash my 9X. It just does not write on the flash memory, here is the log when I try to flash it:
```
/usr/local/bin/avrdude -c usbasp -p m64 -P usb -U flash:r:/var/folders/zh/1zdrjtmx2_7gddxmx9yxqmdw0000gn/T/1968-flash-check.bin:r
=================================================================================

avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9602 (probably m64)
avrdude: reading flash memory:

Reading | ################################################## | 100% 16.91s

avrdude: writing output file "/var/folders/zh/1zdrjtmx2_7gddxmx9yxqmdw0000gn/T/1968-flash-check.bin"

avrdude done.  Thank you.

=================================================================================
/usr/local/bin/avrdude -c usbasp -p m64 -P usb -U eeprom:r:/backup-2016-01-24-193815.bin:r
=================================================================================

avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9602 (probably m64)
avrdude: reading eeprom memory:

Reading | ################################################## | 100% 0.76s

avrdude: writing output file "/backup-2016-01-24-193815.bin"
avrdude: can't open output file /backup-2016-01-24-193815.bin: Permission denied
avrdude: write to file '/backup-2016-01-24-193815.bin' failed

avrdude done.  Thank you.

=================================================================================
```

Trying to flash the firmware with eepe works flawlessly, what is happening? I hope I could get the Companion to work, at least to change the splash screen. Thanks!

kilrah, 2016-01-24:
`avrdude: can't open output file /backup-2016-01-24-193815.bin: Permission denied`

Looks like you might have automatic backup enabled in companion but set to a folder where avrdude doesn't have write permissions.

GilDev, 2016-01-24:
Well you were right, the backup folder wasn't set at all. Thanks a lot!

closed 2016-01-24 by GilDev