opentx/opentx #3252: glitches when using sbus trainer input

KIrill-ka, 2016-01-24:
When trying to feed sbus into serial port (trainer port mode = Master/Battery) I get short glitches in channel monitor.
Configuration info:
 taranis (first version)
 opentx 2.1.7
 8 trainer channels are configured as output channels

After adding some debug printfs I found that glitches are caused by USART overruns.
It doen't happen if the internal module is disabled. External module in PPM mode also doesn't cause the glitches.  SBUS_MIN_FRAME_SIZE is 23 bytes which allows processSbusFrame to accept broken 25 byte frames.
Please, consider at least some error handling to prevent processSbusInput from processing bad data.

KIrill-ka, 2016-01-24:
Tried to increase irq priority "NVIC_SetPriority(SERIAL_USART_IRQn, 6)" in serial2_driver.
It doesn't seem to help.

kilrah, 2016-01-25:
Duplicate of https://github.com/opentx/opentx/issues/2999

closed 2016-01-25 by kilrah

ledvinap, 2016-08-05:
Most probably fixed by https://github.com/opentx/opentx/pull/3410