opentx/opentx #3257: SWR not working on Taranis Plus version A02

plpetrov, 2016-01-28:
Hello,
After upgrade from OpenTX 2.0.19 to OpenTX 2.1.7 I have lost SWR readings.
Automatic telemetry discovery or manual adding gives no result. No value is displayed.
According to Frsky website (http://www.frsky-rc.com/BBS/viewtopic.php?t=5350) the version of the Taranis Plus A02 requires a radio module firmware update in order to receive the SWR functionality. This update was done and no issue with firmware 2.0.19 were detected. The value for SWR was 0. In firmware 2.7 I do not get any value at all.

Kind Regards,

Plamen

tomkfly, 2016-01-28:
I believe A02 had the hardware upgrade, but not the Firmware, so needed to upgrade the Firmware.

plpetrov, 2016-01-28:
This is the statement of Frsky:
Xu@Frsky wrote:
Hi, yes, we have fixed the SWR in the latest production of the Taranis Plus. You can tell the difference according to the series number on the rear cover which installs on the RF slot , such as :X9DP004213B01 . 
Focus on the last three numbers , if it says B01 , then it is the latest batch with SWR function. 
If A02, then it can be upgraded the firmware to get SWR function . If A01, It can not have SWR function .

kilrah, 2016-01-28:
That's the point, it OpenTX doesn't display the SWR it means the module sends it the version code that says it's not supported. So it's really not our problem...

plpetrov, 2016-01-28:
Firmware of the radio module has been done. In version OpenTX 2.0.19 I get SWR =0. I think that you assume that if the hardware version is A02 then there is no SWR. That is not correct. For this hardware version the check should be based both on the hardware and firmware version of the radio module.

kilrah, 2016-01-28:
That's exactly what we do. We have no way of knowing the radio's version, only the module.

plpetrov, 2016-01-28:
Frsky have announced <<Focus on the last three numbers , if it says B01 , then it is the latest batch with SWR function. If A02, then it can be upgraded the firmware to get SWR function . If A01, It can not have SWR function. >>  Logical conclusion is that A02 with upgrades module firmware has SWR. It worked in version 2.0.19 for me. I tested this once again on my radio this evening and I got SWR=0 using OpenTX 2.0.19.

Should I ask Frsky to confirm their statement again or I should better improve my French ...

bsongis, 2016-01-28:
Why do you say it works in 2.0.19? It seems to me that this SWR=0 shows there is an hardware issue on your module! This issue is now "hidden" in 2.1 because we check the module version.

kilrah, 2016-01-28:
If your SWR never moved from 0 on 2.0 then it is not working. It should fluctuate a bit when moving/covering the antenna. 

If it moved on 2.0 (and consistently, not randomly as that is one of the symptoms of the defective modules) then FrSky would need to update us on the version changes.

plpetrov, 2016-01-28:
It is my understanding that SWR=0 is quite preferable value to have and SWR=255 is the one in the problematic hardware version.

However this is my last post here. For the last several months I have been handling software defects and cases as an interface between the customer testing teams and a software vendor. As I now live in a French speaking country I would really prefer to focus on my French and not on another buggy peace of hardware or software.

This is a quotation from:

http://frskytaranis.forumactif.org/t1489p15-retour-du-swr-sur-les-taranis

Quand cela affiche 255 en permanance, c'est que c'est une Taranis+ A01 ou A02. Donc pas de mesure SWR. Sinon 0, c'est parfait.

slowly5, 2016-04-07:
SWR can work on A02, if you enable in the the source code and compile the firmware yourself it works fine.
I believe/suspect the issue is the module ID function was introduced in B01 hardware.