opentx/opentx #3261: STICKY misses quick switch flip

lshems, 2016-01-28:
If I make a logical switch for Sticky on SAup as trigger on and !SAup as trigger of, a quick movement from SA- to SAup and back doesn't trigger LS.

tomkfly, 2016-01-28:
That Will be because there is a deliberate delay on centre position of 3 position switches.This was introduced to stop triggering when passing through. The default setting is 150 milli seconds, but it can be adjusted in General Settings between  0  and 1 second.

fraternl, 2016-01-28:
Good catch...
I didn't even know it was in there. I will now make sure this is lowered a bit in my models.
And I will also check my launch sequence.
When was it introduced?

tomkfly, 2016-01-28:
Not sure, but it has been there for quite some time.

kilrah, 2016-01-28:
2.0.0 likely.

lshems, 2016-01-28:
Nop, this had no effect, i have it on zero always.

tomkfly, 2016-01-28:
I just realised I read the initial problem wrong and read it as a quick move to "-".

tomkfly, 2016-01-28:
I've just checked on Companion 2.1.7 and can confirm the problem.

fraternl, 2016-01-29:
A "too sticky bug"
Hopefully it doesn't stick in there too long.
I can't check as I don't have radio with me.

projectkk2glider, 2016-01-29:
STICKY logical switches are evaluated every 100ms. If the switch on state is too short it won't be seen. It works as designed.

tomkfly, 2016-01-29:
The switch is switching on with a quick operation and then not switching off with the switch left in the off position.

fraternl, 2016-01-29:
> STICKY logical switches are evaluated every 100ms. If the switch on state is too short it won't be seen. It works as designed.

In my correspondence with him he wrote that there was a clear difference between 2.0.x-behaviour and 2.1-behaviour
This was before he (almost) analysed the problem.

I can imagine that the complexity of the code has an influence on timing. Maybe there where it shouldn't?

@lshems Now that you made an isolated test, can you repeat it on the several radio's you tried it before?

lshems, 2016-01-29:
If it switches on, it should switch off.

fraternl, 2016-01-29:
I'm afraid things aren't actually tried but are dismissed beforehand given a possible reason for the behaviour.

I'm not sure it is now. I still can't test it without my radio. I did see this happen with several of the bugs I posted.

To avoid this, you (@lshems) should make the amount of effort needed to actually try this on a radio as little as possible and seduce the devs to try it.

projectkk2glider, 2016-01-29:
I missed one important point from the whole conversation. **The EDGE switch does not turn off**. I was thinking that you don't see the ON event. This is clearly a bug.

projectkk2glider, 2016-01-30:
I have found the reason for this and can fix it, but before that we need to decide what kind of Flip-flop we want to implement!

Using terminology from https://en.wikipedia.org/wiki/Flip-flop_%28electronics%29#SR_NOR_latch we can see that we can have three behaviors when both SET and RESET inputs are ON:
 * Q = 1 (1,0) – referred to as an S (dominated)-latch
 * Q = 0 (0,1) – referred to as an R (dominated)-latch
 * Alternatively, the restricted combination can be made to toggle the output

Lets have an example:
 * setup `LS1     STICKY     SAup    SBup`
 * LS1 is off
 * SA is down, SB is down
 * move SB up (LS1 stays off)
 * move SA up for 1 second and release it, while keeping SB up

Now, current behavior is: **LS1 turns on and stays on**, to turn it off you first have to move SB to some other position and then back to up position.

New behavior could be (based on the three choices above):
 * **LS1 turns on for 1s and then turns off**
 * **LS1 stays off the whole time**
 * **LS1 toggled every 100ms for 1s then turns off**

I any case, whichever change we do it will change the behavior of the EDGE switch and might bite some users.

@bsongis @kilrah @lshems @tomkfly @fraternl  What do you think, which behavior is preferable.

kilrah, 2016-01-30:
It is all as designed and IMO should stay that way.
@projectkk2glider 's explanation of the sample time that was shot down is the correct one to explain the current behavior.

A real flip-flop has a clock input, and inputs are sampled on a clock edge. STICKY doesn't behave that way (which is the reason why we don't call it a flip-flop), it works on the transition of the inputs themselves. It turns on when input A has a rising edge and turns off when input B has a rising edge, completely ignoring the state of the other input at this moment.

With your choice of inputs and a fast move input B never sees a rising edge because your switch flip combined with the 100ms sampling period is too short for SA- to actually become false before becoming true again.

fraternl, 2016-01-30:
I agree with Kikrah that behaviour should not change.

Is it confirmed btw that there's a change of behaviour between 2.0.x and 2.1.x ???

I don't think I have any issue with how it is working now.

kilrah, 2016-01-30:
> Is it confirmed btw that there's a change of behaviour between 2.0.x and 2.1.x ???

No, it behaves the same in both versions for me.

lshems, 2016-01-30:
The problem with the current behaviour is that it is time dependent. That is killing 'script based' state changes, based on multiple linked logical switches, because they become unpredictable.

Whatever you do, please make sure the speed with which you do things, is not influencing the outcome. Otherwise you end up with impredictable results on linked switch sequences.

lshems, 2016-01-30:
On my taranis plus, all is nearly unnoticable, on my old taranis, it nearly always 'skips' the off state.

lshems, 2016-01-30:
By the way, it works as I expected it for SH and SF, the two state switches. So it is either those who are buggy, or the others.

griebd, 2016-01-31:
Just a question, why would I move a switch (please note, not sticks but switches) faster than the minimum timing need by the firmware/hardware to read that change? (that is what I'm understanding as the issue here, if it's not the case, please delete this comment altogether)

lshems, 2016-01-31:
That's exactly the problem. You don't.

So if you do it by accident, it should not change anything. Now half of the
switch responds, and the other half doesn't. So it does change something
you don't want or intend to.

viperfan7, 2016-01-31:
I ran into the same issue, to fix, using your example of SA, I set the AND switch to be SA<State>, meaning it will only be on if the switch is actually in the desired state, works quite well and it also prevents the SA- state from activating on a fast throw from SA^ to SAv, doesn't fix it for no state, but it works for logic functions

kilrah, 2016-01-31:
Why use an LS for that? Just do it in the mixer as usual. 

[image]

lshems, 2016-01-31:
OK, attempt to support with a use case:

I use a source SA on input1
I make a LS1 with input1=100, corresponding to SA switch down
I make a LS2 with sticky on !LS1 ; LS1 

Now I would expect LS2 to be active when SA is in the middle or up position, and inactive when SA is in the down position.

If I change the source on input1 from SA to SF, the LS2 would still be triggered when SF is not in the down position.

So by only changing the source of input1, I can change from a three state switch to a two state switch, maintaining the same functions triggered on the LS2.

This is useful for templates where people use a two or three state switch for controlling launch sequences, and you don't want to explain people how to change things on different places in the program if they want to use a different switch. Because they will simply not understand it.

You can explain them they only need to change the source on input1.

Of course you can do this in other ways, but I don't think in a way where you would change the switch assignation in one place only.

Any suggestions to solve this are welcome, to prevent being forced to use sticvky for this. But not with more LS, because I don't have any left.

Apart from this use case, I would expect the trigger behaviour of any LS to be independent of the duration of an LS input, apart from edge, which is designed to use the speed to define a trigger. I'm not referring to the duration of the output, which is off course linked to the duration of the input, and cxan be delayed or limited in duration by the appropriate settings.

kilrah, 2016-01-31:
See my last line above.

Input1: 100% SA
LS1: a<x Input1 1

Done. Change Input 1 source to SF, behavior is as desired.
You save an LS in the process, and don't use inappropriate time-dependent things.

viperfan7, 2016-02-01:
Kilrah is right, I just don't see how what you're trying to do can't be done with inputs alone and be done far easier.

lshems, 2016-02-01:
OK, full use case instead of simplified one:

launch sequence for two-pos temp switch:
1. pull the launch switch: launch mode
2. release the launch switch: zoom mode
3. when flaps up and elevator > 10%, end zoom mode and return to normal flight mode selection on some other switch

launch mode for three-pos switch with pull-side spring return:
1. pull the launch switch: launch mode
2. release the switch (to MID pos): zoom mode
3. when flaps up and elevator > 10%, end zoom mode and return to normal flight mode selection on some other switch
4. push the launch switch: end zoom mode and return to normal flight mode selection on some other switch
5. return to mid-pos: stay in normal flight mode selection.

Now, i cant figure out how to do this in another way than with sticky, and allow the user to choose which switch to use as launch switch by changing only 1 setting?? In my setup it is only input17.

I have:
[image]
where LS18 triggers launch, and LS19 triggers zoom, and flightmode logic is setup up so that if LS18 and LS19 are not triggered, other flightmodes are active based on someother switch position.

projectkk2glider, 2016-02-11:
@lshems what is the purpose of L23? 

I will be doing my DLG setup soon and I will post my solution here.

projectkk2glider, 2016-02-15:
I will close this since this is the intended behavior. This is how the STICKY is working:
 * when in OFF state: 
   * Ignore V2 changes. 
   * If transition from 0 to 1 is found in V1 signal, turn the switch ON.
 * when in ON state: 
   * Ignore changes in V1.  
   * If transition from 0 to 1 is found in V2 signal, turn the switch OFF.
 * Notice: the VX state is sampled/tested every 100ms!

closed 2016-02-15 by projectkk2glider

lshems, 2016-02-16:
OK