opentx/opentx #3263: Improvement: Make channel order useful again

lshems, 2016-01-30:
Why not change the servo screen to allow channel assignmenet per output. renaming servoscreen to Outputs would be logical as wel.

Screen mockup here :

[image]

mhotar, 2016-01-30:
There is something under the hood. But I am not sure what is it exactly.
I would see the channel order fixed here in the "Outputs", so It looks the same as the real receiver hardware. It will be easier to understand to it. But it will be nice to enable to assign "Surfaces" to them. By "Surfaces" i mean to remove mixer line assignment to Channel, so we will have real virtual channels "Surfaces". It will do exactly the same you are asking for, but will give us more potential. We can then use one "Surface Name" through "Mixer" and "Outputs" screen.

I would personally like (some time in the future) to have "Outputs" and "Mixes" joined to one page called somehow like "Control chains"  But I do not think you are ready to accept It yet :)

kilrah, 2016-01-30:
> renaming servoscreen to Outputs 

That's been done long ago.

I am not really keen on yet one more mapping layer, we've got enough of those to be confusing already...

lshems, 2016-01-30:
Agree.

Why not as a compile option?

Same goes for reassignement of input switches, the physical ones. Similar problem/request/feature.

for the real die-hards, they want complete control over the physical inputs of the real world to opentx, and complete control over the physical outputs to the real world. 

HARDWARE mapping. Start has been made already.

If I wanted to assign a physical switch to the rudder stick input AS HARDWARE MAPPING, why not??

kilrah, 2016-01-30:
> If I wanted to assign a physical switch to the rudder stick input AS HARDWARE MAPPING, why not??

Because it's useless?
That's exactly what the Inputs layer is there for.

mhotar, 2016-01-30:
I should agree with Kilrah, to add another layer to the mix to channel connection will add more confusion than profit. The cost of any change in GUI principles is too big in the situation where we do not have team of the documentation writers and video creators available to incorporate any change we have done.

lshems, 2016-01-30:
Look, I'm just suggesting things. I find it rather discouraging to have discussions where nothing is pushed, being stopped with terms as useless. Useless is in the eye of the beholder. I see use cases where it would help, you don't. 
Thats fine with me, as long as the discussion is without judgement, and about opinions.

text is already a difficult medium to express opinions, so instead of writing 'it is useless', it would be more polite and respectful to write 'i think it is not useful'.

And opening your eyes is what the reader of the suggestion I made also needs to do:
nowhere do I state that it is neccesary, and that things cant be done already.

back to technics, but please try to be nice to each other ( and to me). We all have good intentions, and an open issues list is what it is, an open issue list.

If you are getting tired from suggestions not being in line with your roadmap, publish it. Then people have an idea on where things are going, and if suggestions are totally out of range, or perhaps adding to the quality of the roadmap.

kilrah, 2016-01-31:
> Useless is in the eye of the beholder

Sure is... but that goes both ways! Some people have spent days/weeks to think and come up with something that seemed like a good balance between the current and potentially future expectations of most users.. and some of your suggestions sound like "throw all of that away and replace it just because 1 guy somewhere might have a use for it one day"... Useful is in the eye of the beholder as well. 

> I see use cases where it would help, you don't.

Then... mention and explain them?

griebd, 2016-01-31:
I'm NOT a big contributor here (quite on the other way around), and I sometimes disagree with some positions @kilrah takes... BUT this time I agree with him!!!

And most of your issues @lshems are of the kind "this way is better than the way already implemented", why don't you complain that there is no good documentation? That is a real issue!!!!

I will take this opportunity to say something I'm thinking later: I'm worried to were OpenTX will go from here!!! Yes, we should have features, yes, we should have a system that does what we RC users expect our radios to do, BUT it needs to be kept  simple (KISS philosophy)!!!! Sorry, but sometimes I feel that OpenTX is loosing its focus...

Back to the point, @lshems, if you have ideas, go to the forum and discuss them with other users! See how they do! There are plenty of gliders (some of the most exigent RC fighters out there) migrating to OpenTX and they are quite happy (me included on the happy part) and they are not asking for anything, they can do I all they NEED and almost all they WANT!!!! (people even play music on the radio, why?! :wink:)

If you really find more users thinking like you on the forums and no solutions, them come open a issue!!! Think about the FREE TIME all those people dedicate to OpenTX and try to help then to spend less time!!!!

bsongis, 2016-01-31:
> Sorry, but sometimes I feel that OpenTX is loosing its focus...
An example? 

Otherwise I do agree, we loose too much time in answering to all ideas / issues!

JivaroFAD, 2016-01-31:
Linking input(s) to channel is what we already do in mixer. If channel affectation is changed on receiver, we can already move mixer line(s) to the right channel. So adding a second linking way like this is usefull and does not improve anything.

But... I suggested some years ago a similar thing, but with a big difference : with a normal/reverse option in mixer for opposite functions (like ailerons), to allow linking two or more channels to only one mixer output. That's avoid to make thing twice.

lshems, 2016-01-31:
I never would even dare to say that what others have created is useless, or my idea is better. If you read my posts, you will notice that they are mostly suggestions to generalise an already existing concept. Therefore, they are not backupped with usecases. Simply, because there aren't any.

The roadmap I'm referring to, that apparently doesn't exist, would try to describe some of these general principles, and mention some that are desirable, but not yet implemented.

Instead, we are looking at a myriad of loose items here in the issue list, where in general the response to a request, if that request is back upped by a use case, is to point out how that use case is already being supported. The request is then rejected.

That is fine with me, I do not own opentx, nor am the only one having requests and ideas. I'm grateful to the guys spending time programming, and admire their persistence and constant flow of output. 

I just feel that there is not enough time to properly reply to all the issues raised. You are buried in your own success, by implementing features at lightning speed.

Just try to canalise things, instead of trying to reply to every issue, even if they are mine. Make the issue list in three parts, suggestions, requests and bugs. Put different priorities on those issues, recategorise issues at will, that's your judgement. This will make people feel heard, instead of ignored.

At least try to see requests in perspective, and don't feel offended or become annoyed because requests are asking things that can already be done in a different way than the requester wants it.

Sorry, just trying to add to the general concept, and getting the feeling it isn't effective. I'm not the first, and its up to you to decide to do something with my suggestion or not.

Really, no bad feelings nor intent to be a nuisance. But please stop to block discussions by referring to the time spent. You don't know how much time I spent helping people with opentx, I don't know how much anybody else spends on prgramming or any other attempt to get forward. Time spent is no excuse nor an argument in any discussion, apart from those about money. (And that's not an issue here, is it!)

kilrah, 2016-01-31:
> If you read my posts, you will notice that they are mostly suggestions to generalise an already existing concept. Therefore, they are not backupped with usecases. Simply, because there aren't any.

> just trying to add to the general concept

OK, but if you have a broader view then make that known, and describe it! This issue for example is exactly the opposite, it's a small thing "hidden in a corner" that you propose to change, but without any explanations as to why and how you'd see it integrate better in the whole system.
Basically I can turn back your comment to you, if you want to make suggestions about the large scale evolution of the whole package then you do need to supply the whole roadmap you'd envision, exactly like you're requesting from us! Only when you do so can we make a picture of where your request sits, and decide whether or not it's a direction we want to take.

> we are looking at a myriad of loose items here in the issue list

This issue being a prime example!
So I do understand your point, but ironically your issues suffer from exactly what you keep telling us is a problem! That's why there's some resistance... If you'd like us to listen, please at least follow your own principles!

>Just try to canalise things, instead of trying to reply to every issue, even if they are mine. Make the issue list in three parts, suggestions, requests and bugs. Put different priorities on those issues, recategorise issues at will, that's your judgement.

We do just that, and closing an issue is nothing but assigning it a priority of 0... More than the time spent, it's the way people are - we are impulsive guys, and when we see an issue it's either it picks on someone's interest / curiosity and is treated quickly, or doesn't and it will linger there forever never to be touched again. That's why some issues are closed quickly, because leaving them open when we know we won't handle them just adds clutter, adds more boring work to go through them in the future making it less likely we ever go and read them again, and... may give false hope to the poster.
If we had more people we knew might pick them up we'd keep them open, but we know each other quite well and we unfortunately very rarely see new contributors come and take some random issues up. Yes there may be more if people knew how we would like them to handle them, but... it's the same impulsivity that is the reason for the lack of written roadmap. For example yesterday I had no idea how we'd be handling telemetry display on the Horus... but through a "flash" we started getting ideas and discussing it last night, it took an hour, now we know exactly what we want and work has already started. We're not the type to spew multiple pages of text of discussion with 10 people over weeks to decide what to do, it's something that just comes when it comes. Similarly, most of the time when we read an issue we have no plan regarding its subject - but we will know straight away whether it's something we'd like or not. The decision is quick, either there's a chance and it stays open, or there isn't and it gets closed. 
If you really want something the general principles are those found throughout OpenTX today, and the roadmap is to keep along with them... 

Also, it's not because an issue is closed that it isn't taken into account. We will typically remember things but just put them away to be treated as part of a larger rework later.

lshems, 2016-01-31:
"Linking input(s) to channel is what we already do in mixer. If channel affectation is changed on receiver, we can already move mixer line(s) to the right channel. So adding a second linking way like this is usefull and does not improve anything."

Sure, I now have my mix lines all shifted to the 17 and up, and have mix 1-16 just refer to those mixes.
Changing mix lines can be done of course, but not in the field when you just need to change order to test another plane with your program and they use another order.

Just a waste of 16 channels, using them only to reorder.

P.s. side effect if done the other way around, as you suggest, is indeed that you could refer to one mix for symmetrical output s. Nice!  That would even be a better option.

lshems, 2016-02-01:
André,

Thanks for your reply, it surely adds to understanding how things are
working out for you guys. And understanding is the first step in
cooperation.

Also, I'm the first to admit that I have my flaws, and some of my
suggestions for you also apply for me, as you have correctly put in you
reply.

You work impulsively, that's the way you enjoy what you are doing, and we
have to thrust you take the right decisions. Also fair enough. When work is
done voluntarily, you cannot expect any garantuee for the process.

In my work I have about the same profile as you guys. I tend to call it
'organic management'.

The key thing here is that it must be clear to the people I work for I do
it that way. Otherwise, it will not work out.

I think there is the difference. You guys don't work for anybody but
yourself, and share your results for those who wants to use it, and take in
suggestions and bug reports to improve it, as you like.

That's at least what I make up from several previous discussions.

That's fine, but you should realise that opentx is becoming a factor of
importance. People are starting to have different expectations. They have
no real alternative.

This brings with it a certain responsibility, which, if not taken up in
some way or another, can bite the whole opentx thing.

I'm not saying you should do anything, but I'm convinced it would be wise
to do so.

At least make it clear on as much places as possible how you work, why you
do so, and what people can expect.

It was clearly not found by me in my extensive search for information on
the 'organisation' of opentx, for which the result was nil, other then the
names of you guys on the opentx organization on github.

Still trying to be helpful,

Guido

closed 2016-02-18 by kilrah