opentx/opentx #3268: [Suggestion] USB Joystick set to 8 Axis and 24 Buttons

viperfan7, 2016-01-31:
Playing around with the USB joystick, I've noticed there's just not enough button outputs to handle every switch state. This greatly limits the possible uses of the transmitter in some sims, since as a joystick, there are no 3 position switches.

To fix, I suggest increasing the button count from 8 to 24, this will not only allow every switch state to be outputted, but will allow every mix to have an output while staying well within the limits of DirectInput.

kilrah, 2016-01-31:
Do you have a particular use for that? Because the common applications for using the Taranis as joystick typically can't even make use of buttons at all.

viperfan7, 2016-02-01:
Just from a usability standpoint, most of the sims I've seen do have the capability to use more joystick button bindings, and this would allow people to properly set up bindings with the TX instead of having to shift from TX to keyboard.

I'm not sure what the limitations on joysticks on Mac an Linux are sadly, but I suspect that they would be around 64-128 buttons as well, I may just try making it myself to see how well it works, would take time as I've never done anything in C++ though

projectkk2glider, 2016-02-01:
@viperfan7 you have given your self quite a task there considering your experience level :wink:

viperfan7, 2016-02-02:
No C++ experience, never said anything about other C based languages, C, some BASIC, quite a bit of C#, even some java and python

viperfan7, 2016-02-02:
Hey, I'm not inexperienced with programming, just with C++ and HID, would take me a while to learn it, but could learn it

viperfan7, 2016-02-02:
I think I may have figured it out, is there a way to build it on windows or will I have to run a Linux VM and build it though Linux?

projectkk2glider, 2016-02-02:
Take a look at the wiki pages. I find compiling/debuggibg on Linux much easier. You can even sniff USB traffic with Wireshark. And look for online tool that generates USB HID description packet for you.

bsongis, 2016-02-02:
In 'next' compilation on Windows is not that difficult!

viperfan7, 2016-02-02:
@projectkk2glider Figured as much, trying to set up a VM with mint right now, too lazy to do a live USB drive.

Tried compiling in windows, only issue was that PyQT4 failed for some reason

commit 0ebbe08f (2016-02-05): Joystick now supports 24 buttons (ch9-32) (fix #3268)
commit cbf14235 (2016-02-05): Merge pull request #3276 from opentx/kilrah/3268_16buttons

Joystick now supports 24 buttons (ch9-32) (fix #3268)
commit 110c5da1 (2016-02-06): Joystick now supports 24 buttons (ch9-32) (fix #3268)
closed 2016-02-05 by bsongis