opentx/opentx #3271: [Feature] Change flight mode though on screen menu

viperfan7, 2016-02-01:
Just a minor suggestion, why not add the capability to change flight mode though the on screen display, perhaps in the overview screen using the + and - keys.

lshems, 2016-02-02:
Make a LS per FM you want to select, set as criteria to be equal to a GVAr,
and use a LUA script to change the GVAR on button pres, or use a trim or
switch as input on a SF to change Gvar.

This is the reason I wanted to have some free GVAR's, independent of FM.
Now you will have to sacrifice a GVAR to do this, having the same value for
all its nine FM dependent values.
https://github.com/opentx/opentx/issues/3246

The + and - are not used in the general view screen, as you have correctly
noticed, so there would be no problem passing them to lua.
So I suggested to pass the + and - to lua, so you can do it exactly as you
want with a telemetry script in "background" mode, so not being "in view".
https://github.com/opentx/opentx/issues/3210

Met vriendelijke groet

Guido ter Horst
+32 499 708000

viperfan7, 2016-02-02:
I tried it with a model script but no success there, didn't try using a background telemetry script though

Is there any way to detect which screen is in view though lua though, so I can ignore input when needed?

lshems, 2016-02-02:
You only get events in the run function, so when your script is active, on
screen.

Nice idea: passing events from one run function to a second script in
background using globals!

lshems, 2016-02-02:
@viperfan7 :

you can only ctach events in teh run function that is on screen for a telemetry script.

My request #3210 was to change that, and pass events to all telemetry background scripts, at least the not used ones like plus, min and enter in the general screens.

viperfan7, 2016-02-02:
I decided that until something like this can be done, best option is to use the throttle trim, I'm not actually doing anything for it as I only use this for simulators, and I wanted a way to toggle between unity based sims and more standard sims that don't have issues with negative values

lshems, 2016-02-03:
Seems a wise decision.

There is some reluctancy to allow the menu buttons to be used in LUA
scripts on any other situation than when a telemetry script is in run mode.
I don't see any reason why not, but they don't see a reason why, so it will
probably be longtime before this will happen, if ever. (i'm still not into
C++ coding myself. Perhaps I get tempted enough to give it a try someday)

I don't see an issue in even open them up for use as flight control inputs,
as there is for me no difference between a button on the control screen and
a switch.

Changing flight modes would be a perfect use case, where you would know
that the buttons are not going to change flight mode when they are blocked
for other use, like menu setup.

they are nicely placed, the plus and minus, under your right thumb.

Actually, there is even now a situation where you use the buttons during
flight: trimming a servo in flight is what I do al the time. LOL.

Met vriendelijke groet

Guido ter Horst
+32 499 708000

kilrah, 2016-02-03:
> I don't see any reason why not

We did mention the reason for our objection - there is currently a clear separation between UI controls and flight controls, and we do not want to start mixing these. 
The effect of the UI buttons should stay constrained to the UI elements that are being presented on the screen and not have hidden effects.

kilrah, 2016-02-03:
Oh and besides that even considering this issue in the first place would not even be possible. Given that flight controls should always have precedence, if you changed FM with the UI it would have to straight go back to what the flight switches set it to anyway.

closed 2016-02-03 by kilrah

lshems, 2016-02-03:
As I mentioned in that same post, that cannot be guaranteed with the
current restrictions either (just by pressing Page long, entering a
telemetry script, from there anything is free. There is no obligation to
show on the lua generated screen what the buttons you press will do.)
That's why I still don't see a reason why not, because there is no clear
separation.

With the introduction of LUA, there is an entire new set of usecases
popping up. Of course it's up to you to support that or not, but it's an
illusion that you can enforce certain use of the LUA scripts, or prevent
others, just because it's desirable or not from an opentx GUI design
perspective.

Your point of view is clear, and understandable, I just don't have the
same, and I think it is normal that I will make my opinion clear to others
suggesting things which correspond to my point of view. In the end, the
line between LUA scripts taking over a GUI role, and the OpenTX GUI will
start to blur anyway.

Met vriendelijke groet

Guido ter Horst
+32 499 708000

projectkk2glider, 2016-02-03:
It is up to the script writers to follow good practices and not use the buttons for flight controls. Or if they decide to do it, at least present the user with the clear explanation/intention of what will happen.

I agree with Kilrah on this one.

lshems, 2016-02-03:
Sure. But the script writers may have other views about safe, logic, clear
etc. You cant force that.

I agree that opentx should be as clear and safe to use as possible. But in
the other hand, blocking things only available to expert users and script
writers is useless and past good practice in my view.

I personnaly don't like certain changes done to the telemetry. Users are
forced more and more to be experts on opentx, instead of supporting safe
use for the mass and expert use for the few.

Tell me which new user will be able to setup receiver voltage low alarms in
opentx 2.1 with a standard frsky x receiver? Let alone fail safe.

Perhaps that doesn't bother you, but I think it's a pity.

Just my cup of tea.

kilrah, 2016-02-03:
> Page long, entering a telemetry script, from there anything is free. 

Yes BUT at least you know whatever the buttons command is handled by that particular script and nothing else, and as soon as you exit it it stops hooking them.

> Tell me which new user will be able to setup receiver voltage low alarms in
opentx 2.1 with a standard frsky x receiver?

The logic switch / special function paradigm for setting an alarm has been standard for 2 years now. Prominent documentation would be more important...

> Let alone fail safe.

No change there, it actually became simpler since the "receiver" option was added and chosen as the default...