opentx/opentx #3273: Bootloader self protect

MikeBland, 2016-02-02:
The bootloader appears to include code in:
void writeFlash(uint32_t *address, uint32_t *buffer) // page size is 256 bytes
that allows the flash memory occupied by the bootloader code to be erased/written.
Given there is at least one report that a flashing failure has left the bootloader not working then writeFlash() in the bootloader should NOT include code to erase or write to the bootloader area.

As originally written, the bootloader also includes a software "blocking" flag that prevents ALL flash writing [by writeFlash()] until a valid "start of firmware" is detected being written to the correct start address. This flag is then returned to be blocking on the detection of:
1. Attempt to write to the firmware start address with data that is NOT a valid "start of firmware".
2. Attempt to write to the bootloader area.

These two items help to ensure the bootloader is not corrupted and so remains available. Even if direct writing over USB fails, the bootloader is still available to flash from the SD card.

fraternl, 2016-02-03:
What I would like to note is that I sometimes flashed a new firmware and within that same bootloader session also read the settings in Companion, altered a few things and then wrote them to the radio.

Because I'm using Companion in "nightlies mode" I get a new firmware offered each time I start Companion. So if I was planning to make some adjustments in Companion I often flashed the new firmware at the same time.
Most of the times without any problems, but sometimes with a dead radio as a result.

I don't know if all the times that happened was also when writing the settings in that same session.

As I wrote in rcgroups, I started flashing the firmware with radio off and then use the bootloader to read/write the settings.

projectkk2glider, 2016-02-11:
After quick check everything appears to be OK, relevant part of the code:
```C
      for (uint32_t i=0; i<BLOCKSIZE/FLASH_PAGESIZE; i++) {
        if (address >= FIRMWARE_ADDRESS+BOOTLOADER_SIZE/*protect bootloader*/ && address <= FIRMWARE_ADDRESS+FLASHSIZE-FLASH_PAGESIZE) {
          if (address == FIRMWARE_ADDRESS+BOOTLOADER_SIZE && isFirmwareStart(buffer)) {
            TRACE("FIRMWARE start found in sector %d", sector);
            operation = FATWRITE_FIRMWARE;
          }
          if (operation == FATWRITE_FIRMWARE) {
            writeFlash((uint32_t *)address, (uint32_t *)buffer);
          }
        }
        address += FLASH_PAGESIZE;
        buffer += FLASH_PAGESIZE;
      }
```
The code above is from function `fat12Write()` https://github.com/opentx/opentx/blob/master/radio/src/targets/taranis/usbd_storage_msd.cpp#L747

MikeBland, 2016-02-11:
That is ONE particular call to writeFlash(), which I know exists. I still believe that writeFlash() itself, in the bootloader, MUST block any attempt to write to the bootloader area or erase the bootloader area.
This protection is in the bootloader as I have it, but was removed during the port to openTx, probably due to making the writeFlash() driver more 'general purpose'.

projectkk2glider, 2016-02-11:
The `writeFlash()` is called only from 3 places:
 * from OpenTX firmware when flashing Bootloader, proper address check is there.
 * from Bootloader:
   * flashing from SD card: start address is always after the bootloader and only gets incremented.
   * in USB mode: from the `fat12Write()` (check performed there)

So I see no problem.

MikeBland, 2016-02-11:
I still disagree, the bootloader MUST protect itself in case of a failure elsewhere.
In addition, the code you mention above is also flawed.
The variable "operation" is static, so is NOT initialised when the function is called. It is NOT guaranteed to be set to "FATWRITE_NONE". If an attempt is made to write to the flash that doesn't complete, then "operation" may be left set to "FATWRITE_FIRMWARE". On a later attempt, this code will NOT stop a write to the bootloader.
If the code in writeFlash() included the protection I describe then this writing to the bootloader would be prevented.

MikeBland, 2016-02-11:
Actually, it is worse than this. "operation" is only set to "FATWRITE_NONE" when the last block of flash is written. Unless the firmware being written occupies the whole of the flash space, then "operation" is left as "FATWRITE_FIRMWARE". So once an attempt has been made to write the flash, even if successful, a second attempt WILL write to the bootloader area!
Please note the TWO items of bootloader protection I described in the first comment. I suggest both of these are re-implemented to protect the bootloader area.

fraternl, 2016-02-11:
I just had it again.
After flashing in bootloader I pulled the USB-cable.
Then I used the menu to go to the normal radio menu. The LCD shuts down, The Power LED stays on (mine has a power-light) and the radio is as good as dead.

I read the firmware using Companion which I will now upload

[Corrupt-after-flashing.zip](https://github.com/opentx/opentx/files/126669/Corrupt-after-flashing.zip)

fraternl, 2016-02-11:
I flashed the radio in power-off state and read the firmware back
This is the result:

[recovered-after-corrupt.zip](https://github.com/opentx/opentx/files/126675/recovered-after-corrupt.zip)

MikeBland, 2016-02-11:
Did you use the bootloader to read the flash, or did you not check the bootloader still ran.
In the corrupt data, the fourth 64K block is erased, but no data written to it (0x30000 to 0x3FFFF).

fraternl, 2016-02-11:
Bootloader was probably corrupt. Tried that before in previous cases and it didn't start.
I didn't try now.
It is working again, now. If it happens again I will explicitly try to start it with trims pushed in.

You should be able to check it by flashing the corrupt firmware.

The computer I'm using with Companion is an old XP-machine. It is slow, but quite stable.

kilrah, 2016-02-11:
Flashed the file and the bootloader is fine. Which makes sense since it's not in this memory range. 

> After flashing in bootloader I pulled the USB-cable.

Did you use the "safely disconnect hardware" thingy?

fraternl, 2016-02-11:
@Kilrah,

I do know this and it is emphasized by the power light. I know I need to pull the battery.
Even though I set the switch to off, it stays turned on.

Normally you would not see this, but because I have this light I can see it stays turned on.

No, I did not use the "remove hardware" before unplugging USB. You wrote it's not needed
I somehow can't find it now, but I once made an entry asking for a button in Companion to eject the USB disks.

kilrah, 2016-02-11:
I wrote that without knowing or expecting you to be using a 15 year old deprecated OS. On Win7 and up it's superfluous, but such problems were common on XP.

fraternl, 2016-02-11:
I fly when I stay with my GF. She has an old faithful XP machine. I used to eject these disks, but stopped doing this lately.

It's strange that I can't find that Github entry. I also searched on rcgroups.
I will start ejecting these disks again.

But why not an eject immediately after flashing the firmware?

fraternl, 2016-02-11:
BTW I have this power light for half a year now...

projectkk2glider, 2016-02-12:
> Actually, it is worse than this. "operation" is only set to "FATWRITE_NONE" when the last block of flash is written. Unless the firmware being written occupies the whole of the flash space, then [...]

Mike, the call to `writeFlash()` is protected by outer if case and the state of `operation` does not matter in this case. Here is the relevant part of the code:
```C
if (address >= FIRMWARE_ADDRESS+BOOTLOADER_SIZE/*protect bootloader*/ && address <= FIRMWARE_ADDRESS+FLASHSIZE-FLASH_PAGESIZE) {
  // stuff
  /*some condition*/ {
    writeFlash((uint32_t *)address, (uint32_t *)buffer);
  }
}
```

projectkk2glider, 2016-02-12:
@fraternl your flash readout clearly explains why it is a bad idea to pull the USB cable without ejecting it first in Windows. FLASH sector 5 (from 0x20000 to 0x3FFFF) got erased, but was only partially written (from 0x20000 to 0x2FFFF), because those sectors were not yet written to the radio.

And that sector is not where the bootloader lives! Your bootloader was uncorrupted and is still working.

fraternl, 2016-02-13:
I had it again.....
This time I didn't "pull the plug".
I waited 10 seconds and then did the "remove hardware" thing.
I then pulled the plug, gave exit to go to the radio menu and it appeared to be dead.
Pulling the battery and replugging it didn't bring it alive.

Turning it on, however, did fire up the bootloader.

I will now repeat the procedure....
[Corrupt-after-flashing-even-with-remove-hardware.zip](https://github.com/opentx/opentx/files/129069/Corrupt-after-flashing-even-with-remove-hardware.zip)

MikeBland, 2016-02-13:
That looks quite bad. The area 0x50000 to 0x5FFFF is mostly erased still, although there is a small block in the middle of it with some data in it. However, I don't like the look of the area just before this, there are some 0xFF bytes that look wrong, and the area starting at 0x60000 looks to have quite a few areas with small blocks of 0xFF bytes.
This suggests that the processor has failed to write the correct data at times.

projectkk2glider, 2016-02-15:
@MikeBland are you satisfied with my answer https://github.com/opentx/opentx/issues/3273#issuecomment-183358606. I think this issue can be closed. The problem @fraternl is seeing is the result of OS sending firmware sectors in wrong order - ie it fragments the file or places it at another location. It is a known problem with the current firmware upgrade implementation and we have another issue open to address that.

bsongis, 2016-02-15:
Right, massstorage is not the right protocol to flash the firmware. This is why we should now use the CLI for that. CleanFlight does this If I am not mistaken.

bsongis, 2016-02-15:
@projectkk2glider Did you find a solution to show the Taranis as 2 (or more) USB different devices?

projectkk2glider, 2016-02-15:
Can't really remember now :smile: Which modes would you like to have at the same time?

bsongis, 2016-02-15:
Always CLI active

projectkk2glider, 2016-02-15:
It is not that easy with the USB lib we use, see answer at the end of the [page](https://my.st.com/public/STe2ecommunities/mcu/Lists/cortex_mx_stm32/Flat.aspx?RootFolder=%2Fpublic%2FSTe2ecommunities%2Fmcu%2FLists%2Fcortex_mx_stm32%2FUM0424%20Composite%20example&FolderCTID=0x01200200770978C69A1141439FE559EB459D7580009C4E14902C3CDE46A77F0FFD06506F5B&currentviews=1570)

I would prefer if user sets the USB mode in radio settings. The decision when USB is connected would be too late.

MikeBland, 2016-02-15:
@fraternl may have a hardware processor problem as the last flash dump he posted appears to have flash blocks that are only part written. By flash blocks I mean 256 byte areas. Since these are smaller than the 512 byte sector size, this problem is not related to "out of order" writing.

I still stand by my statement in the first post, regardless of whether higher level code protects the bootloader from being written.
"the bootloader should NOT include code to erase or write to the bootloader area".
This seems obvious and fundamental to me, and was in the bootloader (is in for the ones I produce) when I wrote it originally, source code here: https://github.com/MikeBland/OpenRcBootloader.

kilrah, 2016-02-16:
> @fraternl may have a hardware processor problem 

That is indeed also my conclusion, especially given it is a recurring problem for him even when changing other components, and nobody else has the same issue.

fraternl, 2016-02-16:
I'm not having any other issue with the radio, though and it always flashes fine using dfu-util
This weekend I will swap my GF's machine for a more modern one which will have a clean OS install.

As Kilrah said... no-one else seems to be having this problems. On the other hand most people don't flash that often and the ones that do (like you) all have a decent modern machine for it.

kilrah, 2016-02-16:
I doubt the machine has anything to do with it, I've used at least a dozen different machines, even some virtual without issues.
Given the slow speed it works at I wouldn't be surprised if the DFU bootloader did a number of transparent verifications and retries, and happens to manage to flash after a few retries yet without saying a thing.

MikeBland, 2016-02-16:
"result of OS sending firmware sectors in wrong order - ie it fragments the file or places it at another location. It is a known problem with the current firmware upgrade implementation and we have another issue open to address that."

This shouldn't be possible, unless companion is writing the data incorrectly. In eepskye, I open the file for read/write, and then write to the file. Since there are NO spare allocation units the OS has no option but to write the file in situ. The OS never has to allocate anything. The only way I can see for the OS to fragment the file, or place it elsewhere, is if companioe deletes the file then writes a new one.

If the OS is writing allocation units in the wrong order, then a small change should be possible to have each flash sector erased when it is first written, rather then when the first 256 bytes of the sector are written.

bsongis, 2016-02-17:
Right, but you cannot prevent the user from using cp or his file manager. This is why massstorage is not the best option for this. But it was the easiest, and it's safe when using companion / eeskype that's true!

projectkk2glider, 2016-02-17:
@MikeBland here is actual example of how the file can be fragmented, but written in the same sectors https://github.com/opentx/opentx/issues/1793#issuecomment-68639092

MikeBland, 2016-02-17:
As I understand it that only it happens if you are using OS file commands, but not if you use eepskye/companion with the file opened for read/write.
Originally I set the 'files' to system, hidden and read-only to help prevent use of the OS to write to them.
I did, of course, choose to use disk drives for the bootloader because 'all' OSs have built in drivers for these.
It was not intended that the OS should be used to write the file.
For the Taranis, the FAT could be changed to use an allocation size of 32K (the size of the EEPROM).
This would prevent the OS from allocating the EEPROM sectors for the EEPROM differently.
The firmware could be mapped to 16 separate files, each of size 32K (may need two directory sectors).
Companion/eepskye could easily detect that, and as each allocation uint could be 32K, the OS would be prevented from allocating these differently as well, besides making it very difficult for a user to use OS commands to update the firmware.
You still have the availability of the drivers on the PC.

kilrah, 2016-02-17:
How is writing to the file from companion/eepske or the OS commands any different?

Both will go through the OS's filesystem handlers and whatever methods of buffering and caching and the various ways to flush these caches it may choose to use.
AFAIK no current OS has any method to write to disk at a low enough level to ensure each byte is written at the exact location you want that is available with standard priviledges for obvious security reasons. 
You'd have to do a `dd` for that. Opening a file, writing to it and closing it through standard functions can result in anything that is supported by the underlying filesystem.

MikeBland, 2016-02-17:
If you open a file in read/write mode, then change a single byte in the file, the OS has only a few choices of how to proceed, e.g.:
1. Simply update the sector/allocation block in situ.
2. Read the whole allocation block, update it, then write this back in a newly allocated block then update the FAT to suit. This can only be done if there is a spare block to use.
3.Read the whole file, update it, and write it back somewhere else, then delete the original. Again, this can only be done if there is a spare block to use.
I don't think, without any spare allocation units, the OS has any choice but to write the file in situ..

If you use OS file commands, then you need to delete the existing file to be able to write anything. In this case the OS is free to allocate any blocks it likes.
This is the difference. eepskye/companion use the read/write mode when there are no spare allocation blocks.

bsongis, 2016-02-17:
I did trace the order where blocks are written when using cp on Linux. It was more than 2 years ago and I don't remember precisely, but there was a shift by 1 in blocks order.

Mike is right when saying eeskype / companion do things properly, they open the file in read/write mode, not write only as does cp (and as is was done in companion in the past and that's why I had to investigate). Read / write forces the OS to reuse the previous block order.

This is why I think massstorage is not the right protocol here.

MikeBland, 2016-02-17:
IF you look to use some other method, then I would make sure there are commonly available drivers, preferably built in, for it. This should apply to ALL likely to be used OS, including older versions.
The whole point of the bootloader was to provide an easy to use method with such commonly avalable drivers.
Perhaps, rather than map the flash directly, we should just map the flash as though it is an Intel hex file, and provide releases in hex form, not binary. Then the address information for the data is built in.

griebd, 2016-02-18:
I don't know how much memory/resources are available so this idea may be worthless... that sad, why not change the boot loader so that it provides a virtual FS over USB? Then the boot loader will be responsible to read this virtual FS after the OS has finished writing to it and transfer the data to the flash in the correct way...
A second, more complicated idea, and probably already investigate (I think @bsongis already suggested it)... is to use some standard serial communication over USB (with a private protocol). That would even allow for new features like radio time corrections by Companion, no?!
Again, just suggestions, brainstorming... :wink:

kilrah, 2016-02-18:
Somewhere around 3 years ago I proposed to use the generic HID, because it allowed arbitrary data transfers that would allow implementing any good bootloader-style protocol AND real-time communication between companion and radio for live setting updates, telemetry display or just about everything we intended to do, AND with the advantage of not needing a driver as that's part of the OS...

http://www.signal11.us/oss/hidapi/

MikeBland, 2016-02-21:
To include COM or HID would likely need a composite USB device as I believe we still need the USB mass storage to allow access to the SD card.
I always intended the flash/EEPROM drive should always be accessed using eepskye/companion, not using OS commands.
Rather than us OS commands to write the flash/EEPROM, the intended use there is to write the file to the SD card then disconnect the USB and use the inbuilt bootloader command to write that file to the flash.
Regarding the reference to te EEPROM being written to blocks in the wrong place, this should fail to write to the EEPROM, and I believe does fail on the bootloader as I wrote it. I have protection flags that block write access to the flash and EEPROM, one flag for each. These only unblock access is a valid "start of memory" block is attempted to be written to the correct place. In the example referenced above, the start of the EEPROM block is not being written to the correct place, so the EEPROM should remain unmodified. If the EEPROM was modified (incorrectly) by this operation then this issue is valid as the EEPROM should be protected from the above operation.

bsongis, 2016-02-21:
Right, the bootloader was protected when I did the test, it was not badly written, it was just skipped

bsongis, 2016-02-21:
But I have done the test a few years ago

projectkk2glider, 2016-02-21:
The EEPROM was of course not written, my example just shows what the OS thinks was written to the device. The validity check of first sector is there an working, both for EEPROM and firmware.

MikeBland, 2016-02-24:
I've just seen Wayne from Aloft has posted regarding a X9D with a failed firmware update that included Taranis firmware on a Plus. The failure corrupted the bootloader as well.
Leaving out all the other comments, I would therefore return to my first post with the simple request that the code in writeFlash() in the bootloader, that allows writing and erasing of the bootloader area of the flash memory, is removed from the bootloader.
Just a small change I think involving conditional compilation for when the bootloader is being compiled.

kilrah, 2016-02-25:
The issue Wayne is about has nothing to do with this, it's that FrSky built firmware that somehow has a corrupted bootloader and flashed that to production batches (unfortunately they don't use our releases, and as such we don't control their choice of compiler and build process). 
The issue was never seen with the matching version of OpenTX or any other.

projectkk2glider, 2016-02-25:
@MikeBland I still strongly believe that we have all cases covered and that the Bootloader can't overwrite itself. I have shown you numerous times the protections that prevent the change in Bootloader part of the FLASH.

fraternl, 2016-05-02:
Ever since my GF has a fast(er) machine with Windows 10 I'm flashing it with the bootloader.
It never happened since.
I will report if it does.

BTW.... Will Companion now be forever slow(er)? It became unusable on that XP-machine. It took more than 3 minutes to load a model.