opentx/opentx #3274: Accx,y,z values different when sent over hub compared to over s.port

nigelsheffield, 2016-02-04:
Sorry if this has already been covered but could not find it in search.
When I send accy accz and accx values over hub I get a decimal p!ace difference compared to if I send over s.port
Ie 
S.port I get 1.7
Hub I get 0.17
Displayed.
I am on 2.0.17 using openxsensor looking to send values for artificial horizon , I can fix this in the sensor code no problem but just wondered if I fixed it for my version would it need changing again for other versions if they  match  values between hub and d16.

nigelsheffield, 2016-02-10:
Talking with mstrens and mikeb last night and acc values should be sent as 3 decimal places.
Hub is then correct where s.port is 2 decimal places.
Openxsensor I am using to send pitch roll and yaw for the instruments display lua script , I have it fixed by sending acc*10 on hub so that both read the same in the script but would be nice to know if this is going to be changed in opentx later or not .
Thanks 
Conversation bellow.
Top
mstrens
Posts: 812
Joined: Fri Dec 27, 2013 7:49 pm
Country: -
Has thanked: 9 times
Been thanked: 86 times
Contact: Contact mstrens
Re: GPS sensor
Postby mstrens » Tue Feb 09, 2016 8:16 pm

I had a look at openTx source.
I found 2 tables where the field ID are defined.
One is for SPORT and one is for Hub.
In the table for SPORT, ACC are defined with 2 décimals while for Hub with 3 decimals.
This explains the results you got.

Now, the question remains : is this difference foreseen by Frsky or is it a bug in openTx?
Kilrah would probably be able to answer this question.
Top
User avatarMikeB
9x Developer
Posts: 11520
Joined: Tue Dec 27, 2011 1:24 pm
Country: -
Location: Poole, Dorset
Has thanked: 82 times
Been thanked: 1995 times
Contact: Contact MikeB
Re: GPS sensor
Postby MikeB » Tue Feb 09, 2016 10:38 pm

According to the spec. for the SPort and hub protocols, both should be sending the Acc values as g/1000, so 3 decimals. Without actual sensors I can't check what they actually send.

Mike.

projectkk2glider, 2016-02-11:
Confirmed, the SPORT specification clearly says both HUB and SPORT sensors report value/1000.

projectkk2glider, 2016-02-11:
Do we also backport fix for this to 2.0.x branch?

nigelsheffield, 2016-02-11:
I would say yes to backport it to 2.0, it makes no sense that using hub give different reading to using s.port.
Not may use the acc values as far as I know so probably why it's never been picked up on, and so would not effect anyone in a bad way I don't think.

kilrah, 2016-02-11:
FrSky have never put out a sport acceleration sensor and I doubt anyone went to the trouble to use the acc sensor, hub and vario to sport to test which is likely why it's incorrect.
However, several 3rd party projects now send acceleration values on smart port (ardupilot, cleanflight) and have adapted themselves to how it currently works. 

Fixing it would be good, but a notice or submitting pull requests with the fix to said projects would be nice, and there will necessarily be a period where it will be broken for some users due to unsynchronised releases.

nigelsheffield, 2016-02-11:
Yes its awkward I realise for other projects already adapted,
Like I said fixing it in my sensor is possible, I just times the hub output by 10 more then the s.port output.
Openxsensor is not currently doing this by default , I change the code every time it is updated.
If it does get fixed users would need to be informed to update both opentx and any effected project to avoid broken acc function.
Thanks...

closed 2019-08-19 by kilrah