opentx/opentx #3275: [Bug][OpenTX 2.1.7]Source trim is not considered as an offset

LapinFou, 2016-02-05:
Hi folks,

Something looks wrong to me in the mixer. The weight is also applied on the trim of the source. This seems wrong.
The testcase to reproduce the problem on OTX 2.1.7 is easy:
CH01: [I1]Ail Weight(+100%)
CH02: [I1]Ail Weight(+400%)
If you add a lot of aileron trim, then the CH02 will shift 4x more than CH01.

I always understood the mixer as shown on the diagram below:
[image]

hrenz, 2016-02-05:
Your diagramm is wrong

[ (Input + Offset + Trim) * Weight ] -->Curve --> Delay-->Slow--> Switch-->Modes

LapinFou, 2016-02-05:
Well, to me, the offset should not be multiplied by the weight.
If you do a kind of double rate (which off course must be done in the inputs) in the mixes, then you will get a very bad surprise when switching from low to high rate...

By the way, does, the following diagram, is correct for the inputs ?
Note: **Mode** means **Flight Mode**
[image]

hrenz, 2016-02-05:
Hy,
Input Pre calculation:
(Stick*Offset)*Weight)+Trim

hrenz, 2016-02-05:
This kind of Mixer calculation is very fine, 
because  you can set verry simple a new range 
Example: From -100 to +100   should be set to +20  to +80
Mid of +20 to +80 is 50 --> set Offset = 50
Range is 60  60/200=  0,3  --> set Weigth = 30%

LapinFou, 2016-02-05:
Thanks for the clarification.
I'm not the author of those diagrams. They are available on this great website:
http://www.rcadev.info/OpenTX%2009%20Inputs%2005.htm

kilrah, 2016-02-05:
> If you do a kind of double rate (which off course must be done in the inputs) in the mixes, then you will get a very bad surprise when switching from low to high rate...

Yep, then don't to it that way!

> According your explanation Mixes and Inputs are not consistent each other. 

They're indeed different, they aren't meant ti be identical

> Well, to me, the offset should not be multiplied by the weight.

It's an offset on the control, if that's not what you want then use the offset in the mixer.

LapinFou, 2016-02-05:
OK. Thanks @kilrah. So the explanation done by @hrenz are correct then.
I need to make my own diagrams and update the french video. :anguished:

closed 2016-02-06 by kilrah