opentx/opentx #3277: Can not change the field "Cell sensor" on taranis plus telemetry screen

DanielG63, 2016-02-07:
When you add "cell" sensor in telemetry screen, you can not change the field "Cell sensor" on the remote control. The display shows " --- " without the possibility of changing. This change can be done through "OpenTX Companion" and after saving the changes on the remote control.

Firmware version 2.1.6 / 2.1.7 (I have not checked in earlier versions)
Hardware: taranis plus ver B, X8R, FLVSS LiPo Voltage Sensor.

kilrah, 2016-02-07:
You likely added the Cells sensor mnaually instead of having it discovered. That won't work.

closed 2016-02-07 by kilrah

DanielG63, 2016-02-07:
Thanks!
I deleted the settings and I made a new scan with "Discover New sensor" and it all worked out.