opentx/opentx #3278: [OpenTx] switch visualization on Taranis

MicheleVilla, 2016-02-07:
[image]

Taranis X9E
OpenTx 2.1.7
Switchs names and arrows visualization  too close sticks box on display page

kilrah, 2016-02-07:
What are those labels? did you define custom names?

MicheleVilla, 2016-02-07:
Yes
in Hardware page I have assigned a name for each switch
For examples number 6 switch on left position is 6Sx

kilrah, 2016-02-07:
OK. That page was designed for 2 characters only.

kilrah, 2016-02-07:
There.

[image]
[image]

commit a9a22628 (2016-02-07): Taranis main view adjustments for 3-char switch names (fix #3278)
commit afddad93 (2016-02-07): Taranis main view adjustments for 3-char switch names (fix #3278)
closed 2016-02-07 by kilrah