opentx/opentx #3279: DSM(X) hack module not working on 9XR-PRO

Grawp, 2016-02-07:
On 9XR-PRO the DSMX module from DX5e is not working on 2.1.x releases while on 2.0.x it does.
There seems to be no signal going to the module. I don't have a scope or logic analyzer at hand right now, however the mean voltage on the signal pin is definitely higher (3.25V) than on 2.0.x versions which may indicate real lack of signal + just having pullup on.

The problem seems to have appeared with a9e64c36036bb6d4ee0fb115c24a920c120938e8 commit. On previous commits the mentioned DSMX module works just fine.

Grawp, 2016-05-07:
I've connected to my 9XR-PRO via SWD debugger and finally I was able to trace the problem down to not setting correct `required_protocol` in `setupPulses` function in _pulses_arm.cpp_. This is caused by adding `defined(PCBTARANIS)` in a9e64c3 commit. Was there any reason for this?
I've removed the define and it seemed to be working for me so I made a pull request #3471 

Btw. If somebody is interested in knowing how to relatively easily connect SWD debugger to 9XR-PRO and use OpenOCD+GDB+QtCreator(with Bare Metal plugin) to step through the code, I could make a wiki page or a blog about it. There are some small changes necessary to be done in Makefile and the code in both _OpenRcBootloader_ and _OpenTX_.

projectkk2glider, 2016-05-08:
Your changes in #3471 look good, I will let @bsongis merge. 

>There are some small changes necessary to be done in Makefile and the code in both OpenRcBootloader and OpenTX.

Please elaborate.

projectkk2glider, 2016-05-09:
The #3471 needs to be ported to the `next` branch also.

Grawp, 2016-05-09:
> Please elaborate.

Do not switch PB7 and PB6 to GPIO via `MATRIX->CCFG_SYSIO`
and since PB7 is used as CARD DETECT you also need to debug without sdcard inserted or physically rewire the CD to some free GPIO & change the code. This must be done both for the OpenTX and the bootloader.
https://gist.github.com/Grawp/7553e529841b96c1b1c9a2511025c99c
Compiling with `-O0 -g3` is also a good idea.

projectkk2glider, 2016-05-09:
Makes sense, I did not know that SWD pins are shared. They are not shared on Taranis. 

And did you make a mistake with `-g3` option, it is not know to me?

Grawp, 2016-05-09:
> And did you make a mistake with -g3 option, it is not know to me?

`man gcc`:

>-glevel
Request debugging information and also use level to specify how much information.  The default level is 2.

And by removing `dwarf-2` I let gcc decide what debugging format to use. I'm not saying the `-g3` is the best option, but I'm really used to seeing it in many many projects.

projectkk2glider, 2016-05-09:
Thanks, I learned something new :+1:

projectkk2glider, 2016-05-09:
Fix ported to `next` in 2df621993eeb8175d3b72374671f77e976a03d5f

closed 2016-05-09 by projectkk2glider

blairfancy, 2016-06-21:
Is this fix included on 2.1 nightly build?

projectkk2glider, 2016-06-21:
Yes.

blairfancy, 2016-06-22:
I m not able to bind with this version on 9xr pro opentx-9xrpro-heli-gvars-autosource-autoswitch-fr-2.1.8N782