opentx/opentx #3284: Seperate debug output from PRINT command output or increase debug window buffer

lshems, 2016-02-09:
when debugging, you get a lot of info, for instance about files being loaded.

This blocks viewing the content of the variables you want to check by using the PRINT command. It has already scrolled away, so you cannot analyse it.

2 options perhaps:
make the buffer on the debug screen bigger (easy one perhaps)
make 2 screens, on for debug info, and the other for results of the PRINT command

Thanks

lshems, 2016-02-10:
Found the CLI option. Will check that out.
#3000

projectkk2glider, 2016-02-11:
>make the buffer on the debug screen bigger (easy one perhaps)

The scroll back buffer should already be unlimited. Are you sure you have missing lines?

lshems, 2016-02-11:
In 2.0 sure. I'll check 2.1

het volgende:

lshems, 2016-03-06:
2.1 as well.

My main problem is file loading being reported. That completely fills my debug window.

mpaperno, 2017-01-15:
A filter as suggested in #3685 would solve it?

mpaperno, 2017-02-06:
Closed via #4385 (and lack of further input).

closed 2017-02-06 by mpaperno