opentx/opentx #3285: inputs lines limited to 32 in LUA call, but available in radio

lshems, 2016-02-10:
When having more than 32 lines in inputs (subs counted as a line) the lines above 32 total count cause LUA to crash when using model.getinput()

both 2.0.19 and 2.1.7 NB

projectkk2glider, 2016-02-11:
There is a bug in Lua interface. We have two numbers:
 * MAX_EXPOS which defines max total lines on inputs page (current value is 64)
 * MAX_INPUTS which defines max number of inputs defined (current value is 32)

The mistake is that we use MAX_INPUTS constant when searching of input. When you request input bigger than 32, you get back a **nil** value. That is why your script crashes. 

The **nil** return will stay, so you have to check it anyway, but the Lua interface should be fixed to enable manipulation of all 64 input placeholders.

lshems, 2016-02-11:
It crashed even if I ask for name of input 2,line 0, if there are already
32 lines used before that on input 1.

het volgende:

projectkk2glider, 2016-02-15:
@lshems please test with the next nightly build. Thanks.

closed 2016-02-15 by projectkk2glider

lshems, 2016-02-16:
Ok.

lshems, 2016-02-16:
Companion 2.1 is OK. Radio not checked yet.

commit d402e2f3 (2016-02-15): Fixes #3285: only first 32 input lines (out of 64) were accessible in Lua
commit 29b8b268 (2016-02-15): Fixes #3285: only first 32 input lines (out of 64) were accessible in Lua