opentx/opentx #3286: Feature request: Expose inactivity alarm

fraternl, 2016-02-10:
Someone just recently wanted to activate the throttle safety if throttle would be off for more than 15 seconds.

Only throttle wouldn't be wise IMHO and I would involve more inputs. Implementing this would take several logical switches and make the throttle safety even more complex.

What he wanted was an automatic throttle safety if he somehow forgot to enable it after flight.

If the inactivity alarm could be exposed as a selectable logical switch I could transparently involve this switch into my throttle safety set of switches.

fraternl, 2016-02-10:
The inactivity could also be exposed as a time value. I assume this timer already exists and it is reset everytime one of the inputs is moved beyond its jitter values.

If this inactivity could be exposed as a value in a logical switch it would be more powerful and independent of the value the user entered for its alarm.

I assume the LUA-script writers are interested in getting hold of this value.

projectkk2glider, 2016-02-11:
Nice idea, we just need to expose the value `inactivity.counter` (which is in seconds). 

@bsongis does this require EEPROM change? or we can add it to the 2.1? The Lua api can be done in 2.1, but the source for logical switches I am not sure.

kilrah, 2016-02-11:
To me inactivity would be out of place in logical switches.

fraternl, 2016-02-11:
@kilrah I have direct use for it.

L1  a~x    Throttle   -100
L2  a>x    Inactivity   60     L1

The throttle safety I have is covered here
If I would have the inactivity timer I could use it to put the throttle safety on if no activity is detected whilst throttle is off. Getting it out of throttle cut is easy... just use the temporary switch....

kilrah, 2016-02-11:
You can do it with a single LS up to 25s.

fraternl, 2016-02-11:
No, that's not the same at all.

opentx-jenkins, 2016-02-12:
I am OK with this change but only in 2.2 in Logical switches. In 2.1 in Lua
is not a problem

kilrah, 2016-02-12:
Should be part of a larger rethinking then, we also have the "telemetry present" and "sensor present" to add

bsongis, 2016-02-12:
They are already in the list, but still not functional if I am not mistaken

fraternl, 2016-09-19:
Can this still be implemented in current 2.2 (next) without an EEPROM change?

JapieG, 2016-09-19:
wouw this is a great idea! I should use it to start and stop SD logging. Only start SD logs if inactivity < 60 or something like that. Great idea!

fraternl, 2016-11-09:
According to @Kilrah this feature needs an eeprom change, but according to @projectkk2glider it doesn't. I'm not able to judge this, although I'm inclined to think it does need this change, but frankly I don't know.

I was told this feature request is put on hold because it needs this change.
It would be nice to know who is right about this EEPROM change.

projectkk2glider, 2016-11-09:
I was wrong, any time we change any kind of list of times (in this case list of LS sources), unless the new item can be added at the end, you mess up existing setups (all items after the new item shift their value/index by 1). 

So normally this kind of change is done between mayor versions and eeprom upgrade takes care of proper translation of indexes.

fraternl, 2016-11-10:
Please also add this to the selectable switches in Global Functions.

Given the moment this ticket was created (february), some other tickets that need an eeprom change (power as unit) and the fact that 2.2 is not yet published as an official release I plea to make this change for the upcoming 2.2 version

3djc, 2016-11-10:
It is not beta anymore, but RC, and it is not a 'release' because of Companion, not the firmware itself.

closed 2016-11-27 by fraternl