opentx/opentx #3287: Telemetry screen corruption after emailing script

EtiennePhillips, 2016-02-11:
Hi

On the Taranis Simulator on Companion v2.1.7, I tried to run the Telemetry Screen 2 Example script (from [here](https://www.google.com/url?q=http://lua-20.open-tx.org/telem2.zip&sa=D&ust=1454880632525000&usg=AFQjCNGwVUbUfP3WKXIymkcu2oYSpIvfJA)) , and it corrupts the screen and all buttons stop responding. I've attached a screenshot...

Thanks
Etienne

[image]

kilrah, 2016-02-12:
The lua script is for 2.0, there have been changes in 2.1 and it needs to be updated. I believe the only thing needed is an lcd.clear() at the start of the run function.

projectkk2glider, 2016-02-12:
Exactly as Kilrah wrote. And you can use long press on EXIT key to go back to the other menus.

closed 2016-02-12 by projectkk2glider

EtiennePhillips, 2016-02-12:
Yep, that solved it... I had tried a long Exit press, and that wasn't working either. Everything's 100% now though :)

Thanks!