opentx/opentx #3290: Model ID already used

fnjlong, 2016-02-15:
When a number of models are configured and bound to receivers, but a gap is left in the list.  If I attempt to create a new model, the model wizard is overlaid with the message "Model ID already used".

My workaround is to move an existing model into the gap and create the new one at the bottom of the list.

Not critical, but a bug nonetheless.

Cheers,
Frank

kilrah, 2016-02-15:
It's not a bug, but a warning that the Receiver number that is set by default to the model slot number when creating a new model is already in use by another model (you likely moved models around).
The Receiver lock feature would thus not prevent this model from being flown when the other one that has the same number is selected, and so you might want to check those numbers and maybe make them unique again.

closed 2016-02-15 by kilrah

fnjlong, 2016-02-15:
An empty slot that cannot be used with the wizard.  I think most users would call this a bug.

griebd, 2016-02-15:
If I remember correctly EXIT will just close the warning and things continue as there were no warnings, but I did not check this recently (and not in every single situation possible)...
About the "black art", I do agree and I am/was working to solve this, there is a open issue (#2187) related to that! BUT as things are build right now there is no visible solution for that yet...

kilrah, 2016-02-16:
The EXIT key is processed by the script and as such the popup can't be dismissed without also closing the wizard.

But you can always relaunch the wizard afterwards.

So yes everything in it this ticket is already part of #2187.

fnjlong, 2016-02-16:
You can only relaunch the wizard later on that model location after 
1) deleting the model just created after exit from the error
2) fixing the model ID issue that caused the problem to start with

If you use a lot of D8 receivers you will need to visit each model changing receiver type to D16 to see which model ID it uses.  Proceed through each one until you find the one with the model ID of the empty slot.  You could then 
a) move the model back into the matching empty slot or 
b) change the model ID so they reflect the model numbers as you go.

I am wondering if users are aware of the model ID not changing when moving models around.  

Hopefully these concerns are addressed in #2187.

kilrah, 2016-02-16:
No you can launch it at any time from the SD card browser.

fnjlong, 2016-02-16:
WOW, so you can.  I never knew that.  Thanks I have learnt something new.