opentx/opentx #3292: X16 instead of D16 in German translation

fraternl, 2016-02-15:
@hrenz 

In the German translation of OpenTX Helle has incorporated some extra code so it doesn't show the D8-mode. I'm not using the LBT-firmware, but people are telling me that D8 is working with that.

It seems that the additional code that Helle wrote a while ago is obsolete.
I don't know if the directive "MODULE_D16_EU_ONLY_SUPPORT" is used, but if it is than it might even be unwanted now.

This additional code is not there in the International version.
He's also calling the D16 protocol  X16
This is confusing as it is not mentioned this way by Frsky.

German version:

#if defined(MODULE_D16_EU_ONLY_SUPPORT)
  #define LEN_XJT_PROTOCOLS    "\006"
  #define TR_XJT_PROTOCOLS     "AUS\0  ""D16-EU"
#else
  #define LEN_XJT_PROTOCOLS    "\004"
  #define TR_XJT_PROTOCOLS     "AUS\0""X16\0""D8\0 ""LR12"
#endif

International version:

#define LEN_XJT_PROTOCOLS      "\004"
#define TR_XJT_PROTOCOLS       "OFF\0""D16\0""D8\0 ""LR12"

kilrah, 2016-02-15:
The MODULE_D16_EU_ONLY_SUPPORT (eu option in companion) is still there but handled in the menu code, his change (or most likely rather omission to follow, I suppose it was initially done that way for all languages bu got changes at some point) actually breaks the LR12 choice.

hrenz, 2016-02-15:
This EU-only-support was  only for ETSI V1.8.1 with Frsky MU10% - Mode in 2015
The D8, LR12 Mode was no longer conform, only D16 (= X16) at MU10%

Now, since Jan. 2016 Frsky has a LBT-Version and all Modes are now conform with ETSI V1.8.1
so D16, D8 ; LR12 is now possible again.

At Companion you can select EU-Mode, 
don´t do it and you have all FrSky-Modes D16,D8, LR12

The old Frsky protokoll names are:
"X16" for all X-Receivers  and XJT-Moduls  (=D16)
"D8" only for D-Receivers and DJT and DHT-Moduls

kilrah, 2016-02-15:
> This EU-only-support was only for ETSI V1.8.1 with Frsky MU10% - Mode in 2015
Now, since Jan. 2016 Frsky has a LBT-Version and all Modes are now conform with ETSI V1.8.1

No, D8 has not been changed and thus is still not supposed to be used, EU mode is still relevant. LR12 is OK.
But the difference is handled in the menus, the language file must be changed to match the international version. 

> The old Frsky protokoll names are:
"X16" for all X-Receivers and XJT-Moduls (=D16)

No, X16 has never been an official denomination. We had chosen it at the very beginning because it felt logical to us but it was changed to D16 when FrSky used that, even before the Taranis was released so it was never used in public.

fraternl, 2016-02-15:
@hrenz 
To create the Dutch translation I've used your translation and the original English and I noticed the extra code then. It wasn't until this thread http://www.rcgroups.com/forums/showthread.php?t=2603978 that I saw this X16.

Allerhop says "My X9D and my X9E offer both the X16 mode. It is a german 2.1.7. OpenTX, _i can not imagine, this is language specific_."

hrenz, 2016-02-16:
Hy,

no Problem to change it  to D16
Think its called X16 since 2 years now and all German users are happy with it.

The MODULE_D16_EU_ONLY_SUPPORT is only at the german language file
(think was from Bertrand for Fa. Engel from Jan 2015 ETSI V1.8.1)
And its In no other European language file

kilrah, 2016-02-16:
> The MODULE_D16_EU_ONLY_SUPPORT is only at the german language file

Again it should never have been there.

fraternl, 2016-02-17:
Read that thread and you'll see it has already been the source of confusion.
Different languages are expected to work in the same way.

commit ce5daa09 (2016-02-18): Merge pull request #3304 from hrenz/patch-12

Update de.h.txt (fix #3292)
closed 2016-02-18 by kilrah