opentx/opentx #3295: Inability to recognize Mass Storage Devices on Mac OS 10.11.3 with 2.1.7

koensayr, 2016-02-15:
I'm having a heck of a time getting my Macbook Air to recognize my Taranis Plus as a Mass Storage Device. It was working for a bit, but seems to have stopped. I've pulled the SD card out of the radio and updated it all to 2.1.7. I'm totally at a loss. About all I can see in the Console log is the following:

`
2/15/16 1:16:42.000 PM kernel[0]: 004200.434185 IOUSBHostDevice@14100000: IOUSBHostDevice::getDescriptorGated: type 0x03 index 2 length 50 completed with 0xe00002ed and bytesTransferred 0

2/15/16 1:16:42.000 PM kernel[0]: 004200.434416 IOUSBHostDevice@14100000: IOUSBHostDevice::getDescriptorGated: type 0x03 index 1 length 2 completed with 0xe00002ed and bytesTransferred 0

2/15/16 1:16:42.000 PM kernel[0]: 004200.434709 IOUSBHostDevice@14100000: IOUSBHostDevice::getDescriptorGated: type 0x03 index 3 length 2 completed with 0xe00002e8 and bytesTransferred 0

2/15/16 1:16:42.000 PM kernel[0]: 004200.438717 IOUSBHostDevice@14100000: IOUSBHostDevice::getDescriptorGated: type 0x02 index 0 length 9 completed with 0xe00002ed and bytesTransferred 2

2/15/16 1:16:42.000 PM kernel[0]: 004200.440650 IOUSBHostDevice@14100000: IOUSBHostDevice::getDescriptorGated: type 0x02 index 0 length 32 completed with 0x00000000 and bytesTransferred 9

2/15/16 1:16:42.000 PM kernel[0]: 004200.441478 IOUSBHostDevice@14100000: IOUSBHostDevice::getDescriptorGated: type 0x02 index 0 length 9 completed with 0xe00002e8 and bytesTransferred 0

2/15/16 1:16:42.000 PM kernel[0]: 004200.443156 IOUSBHostDevice@14100000: IOUSBHostDevice::getDescriptorGated: type 0x02 index 0 length 32 completed with 0xe00002ed and bytesTransferred 9

2/15/16 1:16:42.000 PM kernel[0]: 004200.443463 IOUSBHostDevice@14100000: IOUSBHostDevice::getDescriptorGated: type 0x02 index 0 length 9 completed with 0xe00002e8 and bytesTransferred 0
2/15/16 1:16:42.000 PM kernel[0]: 004200.443483 AppleUSBHostCompositeDevice@: AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe00002ed)
`

I found this [link](http://openrcforums.com/forum/viewtopic.php?t=6342) which suggests there is some issue around resource forks. I played with this theory and got nowhere.

Any help would be appreciated, perhaps a new bug?

projectkk2glider, 2016-02-15:
Did it work in any of the previous versions of OpenTX? Did you change/upgrade your MAC? Try another USB cable, or other USB port, etc...

koensayr, 2016-02-15:
Yeah, I've tried three different cables, and another MacBook Air. The OS version on the mac doesn't seem to matter.

The last time it did work, it was connected to my Thunderbolt Display through its USB port. The display was connected to the MBA via Thunderbolt cable. I'll be able to verify that when I get to the office tomorrow. However standalone it doesn't seem to be functioning. On both laptops with none of the three cables I have, including a brand new Belkin one.

projectkk2glider, 2016-02-15:
It could be USB 3.0 issue. If you can, try using USB 2.0 hub to connect the radio.

koensayr, 2016-02-15:
Hmmm... I'll give that a go, going to try and dig one of those up.

Anything interesting in the console logs?

projectkk2glider, 2016-02-15:
I can only see that it is unable to make USB device connection. You could also try to connect the radio **powered off**. In this mode the CPU's built-in bootloader (DFU mode) will be activated. If that one also causes similar problems then the problem is on your side - ie your computer/OS.

koensayr, 2016-02-15:
I tried it in DFU mode with a negative result as well.

projectkk2glider, 2016-02-15:
There you have it then. The DFU mode was written by the CPU manufacturer and should work in any case. So the OpenTX Bootloader is not the problem. The problem is probably with USB 3.0 port. Some users are also having problems on Windows/Linux when using USB 3.0 ports.

projectkk2glider, 2016-02-15:
Another possibility is that your radio has a physical problem, like bad solder on USB port. This will then mean that it won't work on any computer.

koensayr, 2016-02-15:
So maybe we have a documentation "enhancement" here? Perhaps suggesting *not* using USB 3.0 ports and going through a hub?

Again, Its not a port problem IMHO. I just tried my *other* Taranis Plus and it does exactly the same thing on two different machines.

Doesn't work at all when connected directly to USB 3.0 on two new (one brand new in fact) MacBook Air, but does work when connected through the hub.

I'd be happy to submit a PR against a Markdown file somewhere discussing this, if you think it would help?

jtaylor2, 2016-02-15:
>So maybe we have a documentation "enhancement" here? Perhaps suggesting not using USB 3.0 ports and going through a hub?

While USB 3.0 ports seem to be a problem for some people, they work find for many others.  I personally have 4 systems with USB 3.0 ports and my Taranis works fine on all of them in Windows 7, Windows 8.1, Windows 10, openSUSE 13.2 32 bit, openSUSE 13.2 64 bit and openSUSE Leap 42.1 so the problem is not USB 3.0 ports  or drivers in general but evidently certain hardware driver combinations.

koensayr, 2016-02-15:
So can we document this somewhere, I'm happy to do the work, I just want to make sure its done in the best place possible so that people can stop pulling their hair out like I did.

kilrah, 2016-02-16:
Already is in the related Wiki page.
https://github.com/opentx/opentx/wiki/Taranis-Bootloaders

Although it's in a "Windows" section which it could probably be taken out of.

projectkk2glider, 2016-02-17:
@koensayr feel free to edit that wiki page, to make it better. Closing.

closed 2016-02-17 by projectkk2glider

koensayr, 2016-02-18:
Updated the Wiki documentation with a troubleshooting section.

kilrah, 2016-02-18:
For the record, are the macs on the very latest OS version?
In the PC world some who had the problems could solve it by updating the USB chipset drivers. Apple might have updated them in their later OS releases too.

koensayr, 2016-02-18:
Yes. Everything is fully updated.

timsenecal, 2016-02-18:
I have seen this issue myself.  2009 mac mini with usb 2.0 hardware sees the taranis x9d fine.  2013 macbook air with usb 3.0 hardware requires a usb 2.0 hub to see the taranis x9d.  both machines are running the same version of mac os x.

koensayr, 2016-02-18:
If you can get me the Vendor ID and such as I've posted, I'll add it to the Wiki of known issues. Also if you have the model of the hub you use. I'll post that too.

kilrah, 2016-02-18:
I believe what you've taken as chipset info is irrelevant, it seems to be an internal hub to which the Bluetooth module is connected. 
BTW I have the same on my mid-2014 15" macbook pro, and have no issues with the radio.