opentx/opentx #3298: getvalue pilot longtitude etc still not working in 2.1.7

nigelsheffield, 2016-02-17:
cant get this to work , has the name or method changed from 2.0 ?

kilrah, 2016-02-17:
https://github.com/opentx/opentx/issues/2921

`pilot-lat` (number) pilot latitude, positive is North 
`pilot-lon` (number) pilot longitude, positive is East

kilrah, 2016-02-17:
@projectkk2glider it's actually missing from the lua_fields_taranis.txt info file

closed 2016-02-17 by kilrah

nigelsheffield, 2016-02-17:
with code bellow i get 596430 distance in the script, if I use the manual method which  is -- out it works as expected. Am i just doing something wrong here?

latitude = val["lat"]
		longitude = val["lon"]
pilot_latitude = getValue("pilot-lat")
    pilot_longitude = getValue("pilot-lon")
    
    --if gpsLock== 1then
    --pilot_latitude = latitude
    --pilot_longitude = longitude
    --gpsLock =2
    --end

kilrah, 2016-02-17:
I don't know, maybe you'd want to display all 4 values to see if they make sense?

You can't assign pilot-lat and pilot-lon, only read them. They set themselves with the first erceived coordinates after a telemetry reset as they always did.

nigelsheffield, 2016-02-18:
OK did a test with this lua bellow and pilot-lat and lon returned zero. I had gps lock and those coordinates showed up and I did a telemetry reset.

--to get gps coordinates on screen in pilot-lat and lon

local function run()
  lcd.clear()
  --lcd.drawText(0,0,"test",0)
  val = getValue("GPS")
	
	if (type(val) == "table") then
  
  lcd.drawText(0,10,( val["lat"]),0)
		lcd.drawText(0,20,(val["lon"]),0)
    lcd.drawText(0,30,(getValue("pilot-lat")),0)
    lcd.drawText(0,40,(getValue("pilot-lon")),0)
   
    end
end
return{run=run}

projectkk2glider, 2016-02-18:
Wrong code, the pilot position is returned in table, see https://opentx.gitbooks.io/opentx-lua-reference-guide/content/general/getValue.html

nigelsheffield, 2016-02-18:
got it thanks.
so like this.

val = getValue("GPS")
	
	if (type(val) == "table") then
		latitude = val["lat"]
		longitude = val["lon"]
    pilot_latitude = val["pilot-lat"]
    pilot_longitude = val["pilot-lon"]
	end

 perhaps an example would be helpful in case anyone else struggles here, I got my code from someone who had done it like I had it and was not even aware there was a problem...his heading to home would have taken him south forever lol..