opentx/opentx #3299: Timer countdown beeps/haptic

Oldgibber, 2016-02-17:
Couple of points, don't know whether they have been covered elsewhere, Apologies if they have.

1. Companion has a MIXES screen, Taranis has a MIXER screen.
2. Countdown on any of the three timers using beep or haptic gives a voice countdown from 30 seconds with the last 10 seconds repeating "10 seconds" every second. It does not matter whether minute call is enabled or not. Same on both Taranis and Companion.

kilrah, 2016-02-17:
1) You see the actual mixer on the radio but see the configured mixes on companion, but OK...
2) "Throttle state" on the radio because it is already under the "Preflight checks" header-
3) same

4) Update voice pack/delete the timer10 file/use "Voice" countdown if you don't like it. The idea was "you are within the last 10 seconds". Pretty much deprecated now, the file isn't in the current pack anymore, but it still works as it always did if you have it.
5) Already fixed in  #3222

Oldgibber, 2016-02-17:
4 seems illogical, if you set the countdown to beeps, you still get the voice for the last 30 seconds.

kilrah, 2016-02-18:
Same, there are 3 files in old voice packs, one for 30 secs, one for 20, one for 10 that will play if they are there. You can just remove them if you don't want them.

projectkk2glider, 2016-02-21:
I think this can be closed.

closed 2016-02-21 by projectkk2glider

Oldgibber, 2016-02-21:
Sorry guys, but there is definitely a problem here. I've only just had chance to explore this. 
1. Just checked the English Amber sound file on OpenTX Downloads for 2.1 Taranis Plus, and Timer10, Timer20 and Timer30 are still there. 
2. If all three are deleted out of the sound pack, then you get bleeps on countdown at 30 seconds, 20 seconds and then every second after 10 seconds as Kilrah states. However, if any of the voice alerts are still in the sound pack you get them instead of the bleeps. Been trawling around and found this has been raised before. #993 with the same frustrating lack of response.  

I'm trying to do a decent set of Docs for OpenTX, all I can put here is either, delete the sound files, or basically don't waste your time with countdown beeps because it doesn't work.

projectkk2glider, 2016-02-21:
Remaining issue is being resolved in #3312