opentx/opentx #3300: Adapt special function "Instant Trim" to choose between sticks->subtrims or trims->subtrims

lshems, 2016-02-17:
When doing a maiden, sticks to subtrims is handy. But when doing minute adjustments for certain flight modes, especially on gliders, this is way to coarse.

It would be nice if you could send the trims to the subtrims instead, and clear the trims after the event, as you can from the servos menu.

It could be an option on the SF Instant Trim.

kilrah, 2016-02-17:
No reason to do it in flight, if you need to be fine you can use the trim keys... I believe you already asked and was already told that...

lshems, 2016-02-17:
No reason for you, for me there is. Never asked it before, and normally I don't let anybody tell me to do or do not do something, apart from my wife. I might listen to good advice though.

I'm sorry if I annoyed you, or if its to much trouble to respond with some respect. I know you told me to grow a skin. I have one.

closed 2016-02-17 by lshems

kilrah, 2016-02-18:
OK, maybe it was someone else, sorry. 

> No reason for you, for me there is. 

Then, for the 10th time - *take the time to explain it in detail* with usage examples, why it would be better than it is now! 
What I understand from your short posting is:
- You have done your maiden, maybe used the instant trim feature, then after landing used the trims->subtrim feature.
- You go fly again, and want to do fine adjustments. Well, you've got the entire trim range at your disposal to simply trim normally, which if the adjustments are minute should be more than enough? Then after landing you may use the trims->subtrim function again. 

So yes, from your description it seems useless, but... maybe a complement of info will change it.

lshems, 2016-02-18:
Again, for the 10th time, you're considering every request that is already
possible in another way as suggested as useless.

True, if you have priorities to meet, those are not prio one. That doesn't
mean it's useless. That's just rude to say.

My general use case is to prevent ANY changes using the purely technical
standard opentx user interface , because people simply don't know how to
use it.

Your general use case is you don't care, people should learn it.

As said, that's your choice, but an ignorant one. Ignorant to all people
not as tech savvy as you are.

I will not stop posting ideas if I think they are useful to me. I will not
stop telling you you are rude and ignorant, if I feel so. I will not stop
trying to convince you to take requests into account that will improve
usability, even if it can already be done in another way.

Still trying to help make opentx a better system,

Guido

kilrah, 2016-02-18:
> I will not stop posting ideas if I think they are useful to me.

I never asked you to stop posting ideas, they're welcome, BUT **be clear about them**.

>My general use case is to prevent ANY changes using the purely technical standard opentx user interface

OK perfect, that's all I wanted to know! Now I know what your goal is with that ticket. Was it that hard? Had you put that in your first post you wouldn't have had that response. 

I spent 2 hours a couple of weeks ago in https://github.com/opentx/opentx/issues/3263#issuecomment-177616192 to explain you that we needed to know the broader context for your proposals and not just scattered small issues we can't understand why you'd want them and where they would fit on their own, and you seemed to have understood, especially given as that's what you had been asking from us too.

Now you do just the same and didn't improve your way to present your suggestions a single bit, so yeah I'll get annoyed since I obviously completely wasted my time. 

Then this doesn't seem like a good way to achieve what you want - IMO given that the instant trim needs to be configured in an SF that needs interaction with the OpenTX UI - or you do it in a script. If you do it in a script, it would make more sense to ask for the trims->subtrim to be made available to scripts so you can invoke it from your custom UI...