opentx/opentx #3303: Timer beeps multiple times to indicate which minute it is

GilDev, 2016-02-18:
Hi,
When I'm flying (mostly quadcopters, sometimes FPV), I have a timer counting up on my screen to know approximately when I should land (when I don't have OSD). I use the option “Beep every minute”, but it would be a lot more useful if it could beep a number of time relative to the current minute. For example if it counted up one minute, it beeps once. Two minutes, beeps twice. And when it comes to the sixth minute, it just resets to one beep so that you can keep up tracking the number of beeps, then two, etc.
I hope I was clear, it would be an incredibly useful feature to have. Hope it gets implemented.
Thanks!

kilrah, 2016-02-18:
What radio are you using? All platforms that support it have that spoken

GilDev, 2016-02-18:
Turnigy 9X, it just beeps once every minute

kilrah, 2016-02-19:
Good idea. Not sure of how much interest it will generate among the developers though, I believe the last time anything was done for the stock 9x platform was a couple of years ago... having to optimize every single byte of memory has kind of grown old...

GilDev, 2016-02-19:
Too bad, it's still a pretty great radio for a very low price…

kilrah, 2016-02-19:
A 9XR-Pro is miles above in capability, doesn't require modding to install a programmer, and only costs $10 more!

Anyway I see you have some programming experience, if you want to have a go you might learn something :)

GilDev, 2016-02-19:
Yeah I guess I need to update my equipment… 9XR-Pro or Taranis? That's the question! I didn't found clear informations on this: is the 9XR-Pro usable as a joystick with the mini-USB port like the Taranis? I'm on Mac and I'm tired of using CRRCSim for it's audio input capability. I want some FPV!
Anyway, multiple beeps sometimes is more discrete than some woman speaking out loud. Maybe I'll try to implement the feature, I took a look at the code already but it looks pretty messy, I'm not sure where to look. ^^ And I'm no Git expert. If I need to publish my code I will not know what to do: branches, testing…

kilrah, 2016-02-19:
The 9XR-Pro could technically been used as joystick but nobody has written code for it yet.

GilDev, 2016-02-19:
Well again, too bad, but I think I will go with a Taranis+ next, and I will look at the code to see what I can do about those timer beeps. Thanks for your answers!

closed 2019-08-19 by kilrah