opentx/opentx #3305: [Enhancement, Usage][Companion ,OpenTx] Inputs and Mixer, add Clear/Set all Flightmodes

483917, 2016-02-19:
I use Flightmodes in all my models 
And when i make new input or mixer, all flightmodes are selected as default and that is good.
But often i only need an input or mix to be active in 1-3 flightmodes
In Companion it would bee nice to have a Clear/Set-all flightmode Button,
then if i want to have a mix active in FM1 only, I click  the "Clear flightmode Button" and
click FM1 checkbox Done. 
Look at pictures below
[image]

And on Radio when input or mixer flightmode is selected, Long press enter could clear/Set-all flightmodes.
Then i select the FMx i want as usual.
[image]

If this is a useless idea just close this issue

kilrah, 2016-02-19:
Maybe an "Invert"...

483917, 2016-02-19:
Yes that would work too i guess

elecpower, 2016-02-20:
plus one supporting request

closed 2020-07-07 by stale[bot]