opentx/opentx #3308: Taranis Firmware Upgrade Problems Mac

salvapost, 2016-02-21:
Hello
I am getting problems with dfu-util 0.7.

Some info:
I want to upgrade my Taranis X9D Radio Control firmware.
dfu-util 0.7 (installed in my MacBook Air and also OpenTX Companion 2.1.7).

When I try to update the new Taranis RC firmware (opentx-taranis-en.bin) I always get this message:
No DFU capable USB device found
“your driver does not seem connected to USB or the driver is not initialised”

My DFU-util driver is installed, I triple check it, About this Mac>System report>Software>Installations
I install Companion 2.1.7
I have a Taranis X9D (prior the Plus)
The Firmware
opentx-r 2940
EEPR: 215

I read that some MacBook Air have problem with the USB (3) so I also try in my MacPro with USB (2)

I find gnumonks.org and the http://dfu-util.sourceforge.net where I find a new dfu-util version (v0,9)
I am not very technical I download dfu-util 0,9 but I don’t know how to install or if it will be the solution.

Any help very appreciate 
Thank you

projectkk2glider, 2016-02-21:
Are you connecting the radio in powered off state?

Show us the kernel log of detected USB device, not sure how to do this on MAC, but on Linux one would open terminal window and type in `dmesg` the last few lines will show USB related events/errors.

salvapost, 2016-02-24:
Thank for your help projectkk2glider
Yes, the radio is powered off.
I will try to post the kernel log tonight, I am not sure how to do it, will see.

projectkk2glider, 2016-02-29:
Reopen if the problem still exists. Or perhaps seek help on forums.

closed 2016-02-29 by projectkk2glider