opentx/opentx #3310: [OpenTX 2.1.7] "broken CPPM input for wireless trainer" (module input)

robustini, 2016-02-21:
I often used a "Fr-Sky TFR4" rx binded with a "Futaba T8FG" as "slave remote" and has always worked fine with OpenTX V2.0.17.
The rx is connected in the external module CPPM input, with "External RF" module enabled (8ch, 25 ms, 300 us).
Now with "OpenTX V2.1.7" I can't in any way to make it work.
Thinking of getting something wrong I downgrade the firmware again and everything is back to work.
Same isssue with "Fr-Sky D4R-II" binded with another Taranis Plus as slave tx, work fine with 2.0.17 but not with 2.1.7.
Where am I doing wrong?
CPPM with 2.1.7 require some special setting?

kilrah, 2016-02-21:
> with "External RF" module enabled (8ch, 25 ms, 300 us, negative pulse). 

Why? It should be OFF.

robustini, 2016-02-21:
Yeah, tried both, with external Module ON and OFF with "Master CPPM" enabled, same result, it doesn't work.

projectkk2glider, 2016-02-21:
The input CPPM signnal detection was changed, see #3179 for explanation. You will have to ground the PPM in detection pin. Please reopen if this does not solve your problem.

closed 2016-02-21 by projectkk2glider

robustini, 2016-02-21:
I've read the solution, thanks, plug a jack stereo in the trainer port and all work fine again.
Anyway there's no other option at the moment?
You think you put everything back as before in OpenTX V2.2?

projectkk2glider, 2016-02-21:
There is a reason that change was made, because using powerful external UHF module would produce ghost trainer readings.

robustini, 2016-02-23:
Anyway if i leave enable the "Master/CPPM" does not work, but with the external module enabled yes, dunno why.
The CCPM output of the rx is connected to the inputs of the external module, so I think that is correct.