opentx/opentx #3311: sd path not working in ubuntu 32bit version 2.1.7

nigelsheffield, 2016-02-21:
cant get sd card simulated so cant check lua scripts or play sounds etc in 2.1.7 in ubuntu

f_open(/home/nigel/Documents/sd/SOUNDS/en/SYSTEM/0160.wav) = INVALID_NAME
playFile("/SOUNDS/en/SYSTEM/0008.wav", flags=0, id=0)
playFile("/SOUNDS/en/SYSTEM/0160.wav", flags=0, id=0)
findTrueFileName(/home/nigel/Documents/sd/SOUNDS/en/SYSTEM/0008.wav)
	not found
f_open(/home/nigel/Documents/sd/SOUNDS/en/SYSTEM/0008.wav) = INVALID_NAME
findTrueFileName(/home/nigel/Documents/sd/SOUNDS/en/SYSTEM/0160.wav)
	not found
f_open(/home/nigel/Documents/sd/SOUNDS/en/SYSTEM/0160.wav) = INVALID_NAME

kilrah, 2016-02-21:
Path might be too long, try moving it out of "documents" to see if it changes something

projectkk2glider, 2016-02-21:
I could be a file permissions problem. Path length is not a problem, mine is longer.

nigelsheffield, 2016-02-21:
Had it working in 2.0 no problem, and I moved it to other locations to try,  I will check permissions now.

nigelsheffield, 2016-02-21:
Right clicked and checked permissions , seemed OK, I set everything to full access and still scripts don't work.
Sound files work ok

nigelsheffield, 2016-02-21:
Found the problem, feeling silly now but telemetry folder needed to be in capitals , on windows it worked without caps so I was thrown out...
SORRY!

kilrah, 2016-02-21:
Yup, linux is case-sensitive.

closed 2016-02-21 by kilrah

projectkk2glider, 2016-02-21:
Yes, the path is case sensitive, only the file names are searched case insensitive. This way we try to simulate the case insensitivity of FAT.

nigelsheffield, 2016-02-21:
Ok, thanks.
Think I've been caught out like this before..
All working now and wow I like the new debugging and telemetry values simulation.
Should make scripting faster and less wear on the USB socket!