opentx/opentx #3312: Timer problem with beeps

Oldgibber, 2016-02-21:
Sorry guys, but there is definitely a problem here. I've only just had chance to explore issue #3299 again following Kilrah's comments, and as it has been closed, I've raised a new issue. Following on from the closed issue, Kilrah stated that timer10, timer20, and timer30 were no longer in the current sound files. 

1. Just checked the English Amber sound file on OpenTX Downloads for 2.1 Taranis Plus, and Timer10, Timer20 and Timer30 are still there.
2. If all three are deleted out of the sound pack, then you get bleeps on countdown at 30 seconds, 20 seconds and then every second after 10 seconds as Kilrah states. However, if any of the voice alerts are still in the sound pack you get them instead of the bleeps. Been trawling around and found this has been raised before. #993 with the same frustrating lack of response.

Either the sound files need updating or the beeps routine needs some editing. 

I'm trying to do a decent set of Docs for OpenTX, all I can put here is either, delete the sound files, or basically don't waste your time with countdown beeps because it doesn't work.

kilrah, 2016-02-21:
Our packs don't have them, the amber pack was made by a 3rd party.

Oldgibber, 2016-02-21:
The Amber pack is on the OpenTX Download page, and it seems most are using it.

projectkk2glider, 2016-02-21:
@kilrah why don't we remove the play of `timer10.wav` in the code? Is it needed?

Oldgibber, 2016-02-21:
Timer20 and Timer30 also cause problems.

kilrah, 2016-02-21:
Someone might like the current behavior, it doesn't cost anything to leave it in.

tomkfly, 2016-02-21:
I prefer it the way it is now.

Oldgibber, 2016-02-21:
Perhaps the most frequent comment about OpenTX is the lack of up-to-date documentation. 

Decent documentation can make it accessible to more people, but the software needs to be user friendly, and frankly this aspect is illogical. 

Hey ho, its been closed now.

projectkk2glider, 2016-02-21:
@Oldgibber **this issue** is not closed, the other one is.

Oldgibber, 2016-02-21:
Realised that, and edited it.

bsongis, 2016-02-21:
@Oldgibber Feel free to help on the documentation side!

Oldgibber, 2016-02-21:
I am, you've got copies. Now got clearance from my editor to put stuff on GitBooks, was somewhat bothered about copyright issues. - another story.

kilrah, 2016-02-22:
> Surely its not difficult to disable any voice countdown if beep is selected.

The voice files can contain beeps too...

mhotar, 2016-02-22:
Right. I like to use timer10 that way.
[timer10.wav.zip](https://github.com/opentx/opentx/files/140977/timer10.wav.zip)

projectkk2glider, 2016-02-22:
So, the conclusion (again) is that everything is as designed and should stay unchanged. Agreed?

kilrah, 2016-02-22:
Absolutely.

closed 2016-02-22 by kilrah

Oldgibber, 2016-02-22:
Still completely puzzled by the responses. 
Surely if one wants flexibility with the timer, then it can be used in the logical switches and the special functions to do just what the user wishes and with whatever wav files the user wishes. 

The setup screen either on the Companion or the Tx provided what I had assumed was a simple, quick way to set up a model. One would assume that if one selects beep one gets a beep, not some hidden wav file or files that may or may not be present on the sound card. Crazy!

Oh well, I think I shall just leave the documentation to say, if on ticking beep, and it works then OK, otherwise don't use it.

kilrah, 2016-02-23:
> Oh well, I think I shall just leave the documentation to say, if on ticking beep, and it works then OK, otherwise don't use it.

Completely puzzled by that response myself - it's not that hard to describe it how it is and is meant to be.

"When set on "Beeps" it will either do simple beeps, or play the custom sounds in files timer30, timer20 and timer10 if present."

Oldgibber, 2016-02-23:
Not that hard to describe.... Really?

Beeps works if no custom sound files, timer30, timer20, timer10. Beeps at 30, 20 10, 9, etc - this is what I would expect.
Set it to beeps, and you get timer30, timer20, timer10, timer10, timer10, timer10, timer10, timer10, timer10, timer10, timer10, timer10, but no beeps. 

Haptic: If the custom files are not present you get haptic at 30, 20, 15(!!!), 10, 9, etc. Why the 15 sec when not in either voice or beeps????

If the custom files are present you get timer30, timer20 but no timer10, and haptic as above with the 15 sec vib as well.

Where's the logic in all that?

Simply shifting the Amber file in the downloads is not really a solution. What about people who simply upgrade from 2.0, and copy new sound files across without deleting the old files?

As I said above if I want to do clever things with the timers then special functions, logical switches and extra voice commands can provide some very clever timer options.

I'm guessing these timer options are from one of the early versions of OpenTX, and haven't kept pace with the changes and improvements to SF and LS. Please, please simplify these timer settings to do exactly what the tick boxes would suggest.

I really do find this very sad, OpenTX is a fantastic program, its got some great features, and generally its very logical and as I have been working through the program in detail, apart from some minor variance in naming between the Companion and the Tx, I've been really impressed with the versatility of the program, but the way these timers work is really very poor indeed. 

I can see new users faced with the   timer30, timer20, timer10, timer10, timer10, timer10, timer10, timer10, timer10, timer10, timer10, timer10 countdown instead of beeps having a long hard took at the system and wondering if there could also be similar bugs in the way the controls are coded and losing all confidence in the program.

kilrah, 2016-02-24:
> Simply shifting the Amber file in the downloads is not really a solution. What about people who simply upgrade from 2.0, and copy new sound files across without deleting the old files?

They are in a bit of a mess even for other reasons.

> I'm guessing these timer options are from one of the early versions of OpenTX, and haven't kept pace with the changes and improvements to SF and LS. Please, please simplify these timer settings to do exactly what the tick boxes would suggest.

Basically yes, and there's already an issue open to improve how the alerts and things are done. It will eventually get done one day, but the current system works and there's lots of more important stuff to take care of for now, so it might have to wait for a while. Remember we're not offering a commercial product, and that work is done by volunteers who may choose to work on things they'd like, get interested about, or personally think needs improving before what you think does.
And as now answered many times we will definitely not change the way it works until then because some would lose something they want, we'd do work twice, and users would get much more confused by successive changes to the same feature.

The best in the meantime s to properly document the way it currently is - as long as there's documentation people can learn and understand those things.

Oldgibber, 2016-02-24:
Thanks for a sensible answer at last. I do realise the way you work, and the pressures you are under, and the need to prioritise.  Its not too difficult to guess what the currently important things are that are occupying much time. 

Your answer also gives me something useful to be able to put in the documentation, to prevent others making the same points I've already made.