opentx/opentx #3313: Telemetry lua script unload and load from memory to allow switching between more scripts quickly 2.1

nigelsheffield, 2016-02-22:
I am running into memory problems when running multiple telemtry scripts, 
Maybe a solution would be to unload each script from memory when switching between screens?
I know this may have an impact on some scripts if they are to be running in background and displayed only when run function is called but maybe a tick box next to to the selection of script to allow it to be cleared from memory or not would help ?
Or allow some other memory allocation for some variables to be safe but this would be a lot more work I expect..

I know I can switch between scripts from the menu but that is slow if I need a certain screen quickly such as the GPS HUD screen if I lose sight of my glider..

projectkk2glider, 2016-02-22:
I kbow that @lshems has a framework that loads Lua scripts on demand. Perhaps you can use it in your case.

dsbeach, 2016-02-22:
Another option is to use pre compiled scripts. The memory savings can be significant.

lshems, 2016-02-22:
`--[[
	Taranis Dynamic Telemetry Loader - Guido, LShems
  
  fileToLoad is the telemetry file to load, 
  active is the current code state
--]]

local fileToLoad = "/SCRIPTS/telemetry/gpshead2.lua" 
local active=false

local thisPage={}
local page={}

local clearTable=function(t)
  if type(t)=="table" then
    for i,v in pairs(t) do
      if type(v) == "table" then
        clearTable(v)
      end
      t[i] = nil
    end
  end
  collectgarbage()
  return t  
end

thisPage.init=function(...)
  if active then
    page=dofile(fileToLoad)
    if page.init then
      page.init(...)
    end
  end
  return true
end

thisPage.background=function(...)
  if active then
    if page.background then
      page.background(...)
    end
  end
  return true
end

thisPage.run=function(...)
  lcd.clear()
  if active then
    if page.run then
      page.run(...)
    end
    active= not (...==EVT_MENU_BREAK)
  else
    lcd.drawText( 15, 2, fileToLoad, 0 )
    lcd.drawText( 15, 20, "disabled", 0 )
    lcd.drawText( 15, 40,"press enter-button to activate",0)
    clearTable(page)
    active= (...==EVT_ENTER_BREAK)
    thisPage.init()
  end
  return not (...==EVT_MENU_BREAK)  --if loaded as a menu page, the return command needs to be there
end

return thisPage`

Especially for Nigel

nigelsheffield, 2016-02-22:
Thanks , I will try that lshems!
Don't know how to precompile scripts but that sounds a good idea , the result I am looking for should be easy for anyone to use, I like the fact that script are so easy to add and change per model so just needing the end user to install a simple file in a directory and select it like now..

nigelsheffield, 2016-02-22:
Lshems , what would I do with that code exactly?
I have been playing with your toolset today and eventually got it working eventually once I found all the settings that needed doing in my model though on  Ubuntu opentx it gave errors on startup and needed realoading.

lshems, 2016-02-22:
Just put it in a file named xxx.lua, and load it as a telemetry script.

It will load a waiting page, when pressing enter it will load your script,
when pressing menu it will unload the script and enter waiting page again.

I posted some other files on my blog for you. Just check your pm on
rcgroups.

Met vriendelijke groet

Guido ter Horst
+32 499 708000

nigelsheffield, 2016-02-22:
Great thanks!

nigelsheffield, 2016-02-23:
ok it worked once I changed the telemetry folder name to capitals, ubuntu does need case to be correct.
Thanks.
This gives me a place to start, I need an easy to follow example like this to begin and then expand on if need be.

nigelsheffield, 2016-02-23:
Did not work lshems, it loaded once, then I unload and load again and it says not enough memory.

lshems, 2016-02-23:
Well, for me there is a HUGE problem with lua memory management in opentx, or perhaps even LUA in general. I'm not saying it is a BUG, but it is absolutely not clear how to manage memory in LUA; (could be that I'm missing something there, but I had to learn it all by myself, so possibly I missed a part somewhere) 

I am suspecting "lost" functions, variables etc. get never cleared by the automatic memory management. What do i mean by that?

Well, suppose you declare a local variable within a function, and then you delete the function. The space used up by the variables don't get freed up, even if they are not accessible anymore, and not "existing" as far as usage is concerned.

I have managed to work around this by carefull programming, and have adopted some techniques to prevent these issues. Explicitly 'nil'-ling variable before you delet a function certainly could help. You can check my blog, but it is a steep learning curve!

What does this all mean to the normal script user/writer? 

Nothing, as long as you don't want to dynamically load and unload code. 

A lot, if you do. Then you need to do careful programming.

Sorry, that's as far as I can help.

For the GPS script, when I have my sensors, I will probably make the time to rewrite the script to meet the design demands to be able to load and unload it dynamically. But, I m still waiting there.

nigelsheffield, 2016-02-23:
Which is the reason I asked for the option in opentx.
The current 2.1 option is good that I can load any script from the telemetry menu, even if something crashes I can load a different one.
It is just that what I can't do is have one script running which is not essential to flight but musing most of the memory but then when I want an emergency screen like the gpshead2 we are playing with I have to go through the menu to get it displayed.
Not sure if there is an answer but maybe one screen available via a custom function to be loaded regardless of memory used by other scripts so it in effect it dumps the other scripts to allow it to run?
Then it could just be put on a switch if wanted...

How would I go about precompliing my scripts? Can it be done in companion ? If that saves memory I can try that next.

dsbeach, 2016-02-23:
Yes, script compilation is available in companion, but you'll have to build your own companion because it is a compile-time option.

If you have have a build environment, simply add a `#define LUA_COMPILER` to the preprocessor directives for the simulator project you will use (typically opentx-taranis-simulator and/or opentx-taranisplus-simulator).  Once that is done, copy your existing lua script to a file named _**script**_.lua.src.  At simulation startup (or reload via F7) the source will be compiled to _**script**_.lua and then the precompiled version will be loaded.

Once you have a compiled .lua file it can be copied standalone to the transmitter.

If you would just like a quick test of the effectiveness of this approach, I'll be happy to email you compiled versions of your scripts.

nigelsheffield, 2016-02-23:
Thank you, I will send a couple of  files to you soon then!
I have not built my own companion as I did not really understand the process and opentx has handled it so far ...
Is there a possible cheat here, like I just download a version a already built with the lua compiler built in?

dsbeach, 2016-02-23:
There is nothing publicly posted (that I am aware of).  I can give you temporary access to Windows or Ubuntu binaries.  Email me directly at david.s.beach@gmail.com for more details.

lshems, 2016-02-23:
Well, even if it is the practical approach, and pre-compiling is almost a neccesity for larger scripts, it will not solve the basic problem.

If your code is not correct as far as the constraints I mentioned, it's not possible to repeatedly load and unload. You'll run into problems.

Personally, i'm still convinced there is a memory leak in LUA on opentx. Just not capable of proving it.

Repeatedly loading and unloading a function will crash always, even if it shouldn't. That is what I am sure of.

nigelsheffield, 2016-02-23:
I sent you an email dsbeach. let me know if you got it .
thanks.

dsbeach, 2016-02-23:
Been an hour and nothing yet, but I did get the email from github.  You can try Skype (dbeach@msn.com) if that is an option.

dsbeach, 2016-02-23:
Just got it.  Closing Skype and replying via email.

lshems, 2016-02-24:
When compiled, the gpsheading script does load and unload correctly
repeatedly with the code posted before.

Just for info

het volgende:

nigelsheffield, 2016-02-24:
Thanks lshems , now I can compile thanks to David I can do that too if needed!
Compiled It now loads along side other scripts no problem..

projectkk2glider, 2016-02-25:
Related to #3318

projectkk2glider, 2016-02-29:
> Well, for me there is a HUGE problem with lua memory management in opentx, or perhaps even LUA in general. I'm not saying it is a BUG, but it is absolutely not clear how to manage memory in LUA; [...]

Barring a bug in Lua interpreter or newlib memory allocator, here is how the Lua memory management works and why it is not repeatable:
 * Lua interpreter makes a lot of small malloc/free calls.
 * That can eventually fragment the heap memory.
 * OpenTX calls Lua garbage collector quite frequently, so any unused Lua variables/functions should be cleaned from the memory. But that also requires a proper written Lua script. Some docs http://www.lua.org/pil/17.html and http://luatut.com/collectgarbage.html).
 * Lua is not the only part of the OpenTX code that uses malloc, so even after the whole Lua environment is destroyed, not all heap memory is freed. This means that each subsequent Lua script will see different heap state. What that means that borderline Lua script will sometimes work, but other times it will fail to allocate enough memory and will not work.
 * I have done some tests on Lua malloc/free and saw that Lua correctly freed all memory that it allocated.

closed 2016-02-29 by projectkk2glider