opentx/opentx #3314: feature request: add option to disable RF module on boot

csurf, 2016-02-22:
Please consider adding an option to automatically disable the RF module (both internal & external) when the radio is shut off. When radio is booted, the RF module will remain in disabled state until user manually enables it via some means (menu option, button long-press, etc). The option would be disabled by default in order to maintain existing behavior, so it would be set based on user preference. 

The purpose of this option is to prevent against accidentally powering up the radio without the antenna connected (also applies to internal 2.4GHz RF module, as many people have performed the 'removable antenna' mod on their taranis). Also, it has the added benefit of being able to power up and work with the radio without using excess battery power and spewing unnecessary RF transmissions.

kilrah, 2016-02-23:
https://github.com/opentx/opentx/issues/1098

projectkk2glider, 2016-02-25:
I thing issue #1098 pretty much cover this issue. Closing as duplicate.

closed 2016-02-25 by projectkk2glider