opentx/opentx #3316: !sf not available for timer 1

nigelsheffield, 2016-02-24:
cant select !sf for timer any more , it goes from !se to !sg on tx using 2.1.7
I reloaded eep from backed up file in ubuntu the the switch is listed as !fm8 and I leave it alone and load the file to tx it works ok.

tomkfly, 2016-02-24:
SF is a 2 position switch so SFdown is the same as !SF up and SF up is the same as !SFdown, so there was is no need for the !  And it was removed.

nigelsheffield, 2016-02-24:
OK, I see, yes I just checked it and sf up works the same as the !sf down did...
I rememeber thinking  there was a reason I had to use !sf down before due to some timing issues but thinking about it again a 2 position switch is open or closed so that must just have been me being dumb lol.

thanks.

closed 2016-02-24 by nigelsheffield