opentx/opentx #3318: Adding compile lua script to companion as option.

nigelsheffield, 2016-02-24:
Seeing as the compiled lua scripts take much less memory then compiling them each time they are run , could it be added as an option so that anyone could use this option without having to build an environment?
It then allows better use of limited available memory.

dsbeach, 2016-02-24:
I agree, and favor just removing the #ifdef LUA_COMPILER preprocessor directives.

If that is not acceptable, we should discuss alternative approaches to exposing this functionality.  I did a little investigation into what it would take to expose Lua in the companion UI and decided I was not up to the task.  Our choices are limited as long as the only Lua environment is in the radio simulators.

nigelsheffield, 2016-02-24:
As I understand it the fact that you overwrite the .lua file is considered a risk and the reason it was not done already.
I dont think it a problem because normally a user won't be using this function so won't have the .lua.SRC file on their system so no scripts will be overwritten.
To make the .SRC file requires that you know about the compiler option and so should also know to back up the .lua file before hand.

And I've not tried it yet but can the .lua.SRC file just be changed back to a .lua file should the worst happen? I can still open the .lua.SRC file to edit and so could just copy this text into another lua file if needed...

dsbeach, 2016-02-24:
Yes.  You can always revert the .lua.src file back to .lua provided that enough memory exists to compile and run in a single pass.  My GPS triangle scripts are too big to run on the radio in just source form.

lshems, 2016-02-24:
I always have two scripts available on my test environment:
file.lua  and  file.luab

the luab is the backup file, and always in plain text.
If i change the .lua file, I can always revert changes by copying the .luab to .lua

The .lua can also be a compiled version..

If I want to debug in companion, i need to run the .lua all as plain text versions, since otherwise important debug info on errors is lost. Therefore I created a batch program in CMD that copies the .lua to .luac, and then .luab to .lua. I can now debug properly.

If the changes in the .lua plain text files are tested, I delete the .luac so I know I need to compile it again.

after finishing debugging, I revert the action, copy .lua to luab (being the new version) and .luac to .lua

All I need to do now is compiling the .luab files that dont have a .lua version and save them as .lua. This last step is tedious and could easily be automated in my opinion in the companion.

Doing it like this, would prevent any possible errors, since if you dont create luab files, there will never be any compiling done, and if compilation is done, you will never lose the original.
And if you dont delete .lua files that have been compiled, the will never get overwritten either.

to summarize: even with compile option activated, only compile some source file, recognized on extension (could also be .lua.src, but I don't like double dots) if there is no file with the same name and the lua extension.

Just a possible approach.

P.S. and please make this work on ANY location, so also files that get loaded using load should be compiled, if possible, or have it act as batch process on startup for the scripts directory recursively.

would speed up any script tremendously by the way.

projectkk2glider, 2016-02-25:
We had this discussion before, but did not come up with the final solution. Here is my proposal:

 * we use the system similar to the way the Python does this
 * Lua scripts in source (human  readable format) always have `.lua` extension
 * Compiled Lua scripts have `.luac` extension
 * when a Lua script is about to be loaded by the OpenTX check if the corresponding `.luac` file already exists:
    * if it does check it's timestamp. If the file is newer than the same `.lua` file then use it. 
    * otherwise delete it and compile the `.lua` file and write the result in `.luac`. Use the new `.luac` file for execution.

nigelsheffield, 2016-02-25:
Sounds fine to me.
Then the .lua is easily editable and not acidentally overwritten and the.luac gets created for use in the tx..

kilrah, 2016-02-25:
That's how I'd see it too.

lshems, 2016-02-25:
Perfect.

dsbeach, 2016-02-25:
It sounds like there is consensus on this approach.  That is a good thing and I'm willing to make the changes.

That said, I want to point out what I view as the potential drawbacks.  My primary concern is the proliferation of .luac files and the impact that will have.

The obvious first issue is that the number of script files will double. Even if compilation is limited to just the simulator (as I think it should be) those files will get transferred to the radio by lazy users and/or the synchronization tool.  They won't be run on the radio, but I'm sure it will lead to more user confusion.

Another negative impact of always compiling every script and then running it is the fact that compilation (as implemented now) removes debugging information. This again will confuse users in that lua errors in companion will be less helpful than they are now.

I would favor an approach that somehow addresses these issues.

One obvious solution to losing the debugging information is to compile, but run the un-compiled scripts.  The memory savings are needed in the radio but not in companion.

I think the more difficult issue is unneeded compilation.  I'm open to any suggestions, but one approach might be to scan the first line of each script for a compile directive as a Lua comment.  That way the ability to compile would be available to everyone, but only on the files where it is desired.

Comments? Is there a better way?

projectkk2glider, 2016-02-25:
>Another negative impact of always compiling every script and then running it is the fact that compilation (as implemented now) removes debugging information.

I was not aware of this. 

My Idea is that the radio itself (and Companion simulator) would have this compilation functionality. So in effect all that would run would be compiled variants of scripts. Like I wrote, the same way is used by Python. I see no problem in the existence of additional `.luac` files. Why would they bother users?

dsbeach, 2016-02-25:
> Why would they bother users?

I guess I'm just anticipating a negative reaction.  Lots of people resist change, and I think there would be some users who would not be happy to see .luac files show up without understanding why.

Also, I thought FatFS had an 8.3 naming limitation. But the test I just ran on 2.1.7 shows the file browser function handles .luac OK.  Do we want to incorporate .luac support throughout the radio?

Compiling on the radio is an option.  I'm fine with that decision if that is what we agree on.  Right now it is only done in the simulator and only via a compile time directive and only for those files that have a matching .lua.src file present.

lshems, 2016-02-25:
In the previous discussion, I suggested to have all files compiled before
run, at radio startup or whatever other trigger. Those compiled files are
not available to the users, so in some systems folder, same structure as
user folder.

Then it could be completely transparent to the user.

Make a debug option to force execution of uncompiled scripts for debugging.

I don't see any use in debug info on the radio.

Lua debug info is for programmers, not for users? Better make a key entry
in compiled scripts who to contact in case of error.

het volgende:

lshems, 2016-02-25:
But all this is better not said, will quickly become a hassle. Keep it
stupid simple. As suggested.

het volgende:

dsbeach, 2016-02-25:
> Keep it stupid simple. As suggested

As in "just removing the #ifdef LUA_COMPILER preprocessor directives"?

If so, it is a little confusing for people wanting to compile, but has no impact to anyone else. All said, it is still the solution I favor unless a better idea comes up.

projectkk2glider, 2016-02-25:
>Make a debug option to force execution of uncompiled scripts for debugging.

I like this idea, this would be radio-wide option. Or even compile time option. So normal users would run compiled scripts, developers would run from the source.

kilrah, 2016-02-25:
The one thing Im afraid of if we go that way is that people start sharing only the precompiled version of their scripts and preventing people who are not "developers" per se but still able to understand and change 2 lines of code if needed (quite a few of those around) from being able to tweak scripts.

projectkk2glider, 2016-02-25:
This can already happen, nothing preventing somebody to share pre-compiled script. I thing the users will soon seed out the authors that do not provide the source of their scripts.

kilrah, 2016-02-25:
> nothing preventing somebody to share pre-compiled script

I know, but it takes effort to do so. If we implement what is said here by default people will be doing it even if not intended and without even knowing they do.

projectkk2glider, 2016-02-25:
OK then we should hide compiled Lua files somewhere.

nigelsheffield, 2016-02-25:
Not sure I fully understand the problem but ,
Being able to debug a script is necessary for both the writer of the script and anyone wanting to fiddle a little with it or make their own simple scripts .
Being able to pre  compile a script so that a large one or indeed number of smaller ones can all run together should also be available to a normal user.
If you want to stop people from distributing compiled only scripts (there is a version of the gps heads up display which you can buy and this is a separate issue and a complex one as to whether or not this sort of thing should be allowed ) maybe require uncompiled script be in the scripts folder and the compiled script gets built and put in a separate folder automatically and won't be run unless the uncompiled version is present but is not actually compiled again until the .lua file changes ...
For debugging its only needed in the companion so that could run normally as it is now?

Anyway, I  used dsbeach method for the first time the other day and it worked for me, I did not get confused , the only thing that made it harder was that my windows 7 PC would not make the .lua.SRC file so I had to make it on the ubuntu laptop....

dsbeach, 2016-02-25:
I think we are starting to get to a workable solution. So I'll make a proposal that I'll be willing to implement and see if that needs more discussion before code changes start.

Add a new checkbox to the 'Radio Setup' configuration options on the transmitter and in Companion called 'Compile Lua Scripts'.   (Does this require an eeprom update or is there a spare bit can be used?).  The option would be available only to firmware versions supporting Lua. It would be appended to the current list and the default value would be unchecked.

If the option is checked, each time the Lua environment is initialized any referenced script would be compiled (if out of date) with the resulting Lua bytecode stored in a shadow directory structure named 'CSCRIPT' with the file extension of .luac.  If an up-to-date compiled version already exists the compile step can be bypassed.  Note that the compile function must be responsible for creating the 'CSCRIPT' directory hierarchy as necessary.

If the option is not checked, the behavior is no different that it is now. (I like that part the best).

Even if the option is checked, all functions for selecting a script will not change.  In order to select a script in Companion or the radio it must exist in the SCRIPTS directory hierarchy with a .lua extension.

Questions for further discussion:

1. Should changing from checked to unchecked remove the 'CSCRIPT' directory? (My preference is no).
2. How should write errors be handled? One option is to just attempt to run the script from source.
3. Is there a better name for the shadow directory hierarchy?  Does FatFS support hidden directories and if so should we hide it?

Regarding @nigelsheffield comments on distribution / sale of precompiled scripts I'd prefer to not try to enforce a 'source only' solution. It would require additional checks in the code.  It also would represent a change in policy given that nothing prevents that now. I have no personal interest in commercializing Lua scripts for OpenTX but I think it is OK if someone else wants that headache.

It is also worth noting that this proposal does not address @lshems suggestion to compile dynamically loaded scripts.  However, it would not preclude someone from distributing pre-compiled scripts that were dynamically loaded.

Comments?

dsbeach, 2016-02-25:
I just thought of another potential downside to this approach.

This solution assumes an accurate clock on the transmitter. I've seen my transmitter clock revert back to 1970 based on GPS time before a valid GPS clock is obtained.

It also might cause some confusion when multiple versions of a script are available with dates that are prior to the current date. If I download and execute version 1 of a script, and then find version 2 with a timestamp earlier than the date i first ran version 1, there would be no way to run version 2 unless I intentionally updated the timestamp.

In my opinion, this is a whole lot more complicated than it needs to be. If we just remove the #ifdef LUA_COMPILER and update the documentation, people that need it and want it will figure it out.

I'm with @lshems 

> Keep it simple.

nigelsheffield, 2016-02-26:
I tend to agree with dsbeach, keep it simple.for most people it's not an issue anyway and if there are large scripts needing compiling to run which will be distributed widely then it's up to the author of the script to decide if they want to make source code available or not for people to edit.
I personally love the fact that lua scripts so far have been easy to follow and edit.

But if it's decided to go with the compiling on the tx option if a newer version of the script is loaded would the tx have enough memory to do this or would it impact on the available size for scitps?
And putting the compiled version in a different directory how would this impact on dynamically loaded scripts? Would it just be a case of the scripts needing to be made to take that into account?

Sorry I opened such a can of worms again here.... But for the simple need I have of wanting to pre compile a script when I need that option simply removing the if def line works, no need to do much at all really apart from a quick note in the docs .

mpaperno, 2016-03-09:
Hi folks, just found this thread via the RCG discussion.

> As I understand it the fact that you overwrite the .lua file is considered a risk and the reason it was not done already.

If you use an IDE/editor which detects external changes in files (doesn't everyone?), this will ruin your day.  Or you have to make changes in one version and copy that over to the execution directory for every test... yuck.  If anything I'd like to see this process simplified, not further encumbered.

> 
    we use the system similar to the way the Python does this
    Lua scripts in source (human readable format) always have .lua extension
    Compiled Lua scripts have .luac extension
    when a Lua script is about to be loaded by the OpenTX check if the corresponding .luac file already exists:
        if it does check it's timestamp. If the file is newer than the same .lua file then use it.
        otherwise delete it and compile the .lua file and write the result in .luac. Use the new .luac file for execution.

IMHO this is overall the simplest and most straight-forward approach (in fact looks similar to what I posted on RCG).  I would only add, for completeness, that if a .luac file exists but no .lua source file, we can still use use/select/load the .luac version.  Some scripts are simply too big to compile on the TX so it would be pointless/bad to even include the source version.

Python creates .pyc files and no one seems to complain (that I know of :) ).  You can't even get small SD cards anymore, so who really cares if there are a few more bytes in "extra" files.  But most importantly, the compiled scripts are faster to load and use less memory.   How can that be a bad thing especially in an embedded environment?

> This solution assumes an accurate clock on the transmitter. I've seen my transmitter clock revert back to 1970 based on GPS time before a valid GPS clock is obtained.

It just needs to compare timestamps of two files.  Why would the onboard real-time matter? Or the next point about multiple file versions... there can only be two file versions, the .lua and the .luac.  They should be in the same directory.

> But if it's decided to go with the compiling on the tx option if a newer version of the script is loaded would the tx have enough memory to do this or would it impact on the available size for scitps?

It doesn't take any extra memory for this (except maybe the actual process of writing the file? That would be a general OTx issue, not Lua specific).  Which is sorta the point... the plain-text script has to be compiled anyway when it is loaded.  If it can be run on the Tx at all, it can be compiled and saved by the Tx just as well.  From an engineering point of view, repeating the same operation over and over (compiling the script on every load) is a waste.

IMHO having new options in Companion or whatever sounds more complicated and unnecessary.  That a user might open a .luac file in their editor and be confused for a bit until they figure out the right file to use is not a big issue IMHO.  People can already distribute compiled source files now, and it is actually worse because they have to use a .lua extension which is actually really confusing to keep straight.  So at the very least, again IMHO, OTx could support loading .luac files directly.

Sorry if I missed some points but I think this covers the main aspects.  

Cheers,
-Max

mpaperno, 2016-03-09:
I did miss one important point.  I think the Simulator should always run the un-compiled script version so as to expose all the debug info for the developer.  The sim doesn't properly model the available memory on the Tx, so a script should always be tested on the actual hardware anyway.  In other words, whether the sim runs the compiled or un-compiled version shouldn't matter (except for the debug info).

-Max

nigelsheffield, 2016-03-09:
I agree too.

lshems, 2016-03-09:
Having fun with self- editing lua script files.

For me not an issue (yet).

jfrickmann, 2016-11-27:
I have just hit the wall. Adding a third telemetry script to my setup, and I run out of memory. Pick any two scripts, but not three. Aside from all of the philosophical discussions above, giving us the option of using precompiled scripts to save memory, would drastically increase the usefulness of Lua scripting.
The setup that I am working on: https://www.rcgroups.com/forums/showthread.php?t=2578134 
Screenshot of the new script that I want to add is attached.
Pretty pleeeaaase, can we have an option to precompile scripts?
[image]

jfrickmann, 2016-11-27:
With help from lshems, I got my scripts compiled with that old version 2.0 Companion build that compiles the scripts for you. But it would be a really nice feature to have in the current version!!

lshems, 2016-11-28:
Question to this commit: what happens to a lua script that is called from within a lua script? Is it also compiled?? and stored in the location where it was found??

Thanks.

projectkk2glider, 2016-11-28:
If you use a mechanism that is available in stand-alone scripts (ie you return a script name from the run function) the script will be loaded and compiled.

If you use other mechanisms (like require, see #3939) then that file wont get compiled. To have such files compiled would require too much messing with original Lua  library code (which we try to avoid).

lshems, 2016-11-28:
Thanks for the quick reply. 

Just to fully understand: so if i use the load(filename) LUA function, it won't get compiled? 

Would there be an option to create a LUAfunction to compile a script? then we could use LUA itself to compile some scripts.

lshems, 2016-11-29:
Just questioning: the compiler settings are now that all debug info is compiled into the scripts when running a normal lua script. 

Would it speed up things/save memory if it would compile without debug info? certinly on the radio that is not normally needed, and that's the envronment where we get the dreaded outof memory messages?

projectkk2glider, 2016-11-30:
> the compiler settings are now that all debug info is compiled into the scripts when running a normal lua script.

Do you have any idea how to remove the debugging info?

projectkk2glider, 2016-11-30:
Never mind I found it. But are you sure this will be a good idea. Sure, the ram usage will be smaller, but what will happen in case of error?

projectkk2glider, 2016-11-30:
Aaargh, the dump produced by OpenTX compiler is **already** stripped, we use:
```C
luaU_dump(L, getproto(L->top - 1), luaDumpWriter, &D, 1);
```
Last parameter `1` turns on debug info stripping.

mpaperno, 2016-11-30:
Looks like your version already strips debug info.. ?  https://github.com/opentx/opentx/blob/92553b65895dd97fb479730643804521e04f04f2/radio/src/lua/interface.cpp#L226

Here's mine, with comment:  https://github.com/mpaperno/opentx/commit/1ef83c69656a4fe3c43214d588fb5a4ab5700d94#diff-f49e438ac760aed50058e8f25e88649fR1615

[Cross-posting delay! I see you found it.  :) ]

IMHO you do want to strip the debug info (not very helpful when running on radio anyway), however when running in simulator it should load full version with debug (so, not the compiled version).

In my version of the compiler, the original script is always used -- so if it was plain text, then debug info is always included at runtime (eg. when running in sim).  This way there's also a choice when running on the radio -- use pre-compiled version with no errors, or use plain-text if debug is needed.

-Max

projectkk2glider, 2016-11-30:
I think current version is more appropriate:
 * user must really want to compile a script (needs to create `.lua.src` file), users unaware of this will not be bothered with new compiled files.
 * you can still develop with sources - just remove corresponding `.lua.src` file

projectkk2glider, 2016-11-30:
And it was the simplest solution to implement :stuck_out_tongue:

lshems, 2016-12-01:
I was talking about the standard interpreter setting in opentx. Could we by default have the interpreter/compiler not include debug info, and only include it when some compile option is set?

Talking about running normal text LUA scripts now.

projectkk2glider, 2016-12-01:
OK I get it now. I have to check if that is even possible.

lshems, 2016-12-01:
OK, I think there is an important conclusion to draw after several days of researching this topic:

The only reason a precompiled script seems to take less memory, and solves a lot of problems for a lot of people, is that the real-time compiler seems to take all debug info into the compiled chunk.

If, as I stated above, we could simply tell the interpreter it doesn't need to store the debug info when compiling a chunk, we don't need any humbug of precompiled scripts.

I base this conclusion on another approach I researched, and that is saving a (compiled) function from within LUA as a binary, which is done by simply writing the string.dump(function) to a file. You can then load this file (which essentially is the same as we try to achieve with this compile option) later again as if you loaded an uncompiled LUA file containing a function.

By doing so, I found that the size of the saved compiled chunk was actually bigger than the LUA text file I started with!!!

So, if there would just be a DEBUG compile option for companion, forcing the inclusion of the debug info into the interpreter, and using a stripped interpreter/compiler mode as default, there is no need anymore for all this complex stuff. all code should run as normal text LUA files without any problem.

I hope I explained it clear enough, and it helps solving a long fought issue with the LUA memory errors occurring so frequently.

kind regards,

Guido

lshems, 2016-12-01:
Off course the transmitter interpreter should be set to "no debug" as default as well.

projectkk2glider, 2016-12-01:
Nice find! I will see if this is doable.

lshems, 2016-12-01:
It would really save a lot of headaches for a lot of people (that I have already gone through :)

jfrickmann, 2016-12-03:
Thanks for your efforts!! If it is just a matter of removing debug info, I would absolutely say Go for it!!! 
Thanks again.

mpaperno, 2016-12-04:
OK folks, here's my take on the plan as outlined by Damjan on [Feb. 25](#issuecomment-188677129).  It wasn't really that hard, I probably spent more time profiling it afterwards.  In the end, I think the gains are totally worth it, but y'all decide.

https://github.com/mpaperno/opentx/commits/lua-jit-compiler

Please read the notes on the the [last commit](https://github.com/mpaperno/opentx/commit/b247e8f580fa2782d97ec117985be5ea8a3b14d7) for most of the details.

As an added bonus, you can now also force-compile any script by selecting it in the SD browser menu and executing it.  You'll probably get a script error, but that doesn't affect the compilation.

I did some tracing of loading time and memory usage with and w/out the "just in time" compiler.  The summary is in the attached spreadsheet.  Bottom line is that loading pre-compiled versions of scripts saves on average ~35% memory and time over plain-text versions.  The memory savings vary quite a bit between scripts based on a number of factors (like how much the code is already optimized).  There is a time hit on the initial compilation, avg ~200%, over just loading the plain-text version (I'm sure this would vary with SD card speed).  There is no extra memory used though.

[OTx_Lua_compiler_tests.xlsx](https://github.com/opentx/opentx/files/629329/OTx_Lua_compiler_tests.xlsx)

Using this version I was able to load and compile a whole [set of large scripts](https://github.com/mpaperno/AQ-OTx-Lua) right on the radio, which would never have worked before (I always had to pre-compile the bigger ones in the sim, which is how this all began for me).  Mostly is just works by reloading the scripts a few times until there's enough memory for all of them to compile.  Especially stubborn ones can be compiled manually as I described above (when you manually execute a script, all other scripts are first unloaded from memory, so you have the full amount to use).

The only thing this lacks is the ability to select a script with a .luac extension from any of the menus (mix/telemetry/function).  So a .lua version has to be present right now, even if it is never actually loaded.  I'm afraid adding this may get a bit convoluted... but I'll take a closer look.

Anyway, if you guys like this change, I'd be happy to submit the PR, and work this into 2.2 branch.

Cheers,
-Max

mpaperno, 2016-12-04:
Regarding debug info...  with:

    Script  AQTelem error: /SCRIPTS/MIXES/AQTelem.lua:235: attempt to index global 'g_fld' (a nil value)

and w/out:

    Script  AQTelem error: ?:0: attempt to index global 'g_fld' (a nil value)

There may be other differences, but this is the most obvious and consistent.

mpaperno, 2016-12-04:
Just to add, from what I've observed, most of the extra memory usage when loading plain-text scripts occurs during the actual loading -- once the loading (parsing) is done, the free memory remaining isn't that different from when pre-compiled versions are used.  Also the init() function (if any) is called during the load, before any GC is done, so with plain-text you may also get less room for that to execute.

-Max

projectkk2glider, 2016-12-04:
I had a short look at your solution and I like it. I am busy right now with a USB issue and will take more time for this after that one. 

I would not be so concerned when the GC is called, because Lua will do an emergency GC every time it tries to allocate memory and fails. It then tries to allocate the memory second time after the GC has run.

projectkk2glider, 2016-12-04:
I would also prefer if we first have this for 2.2 branch first and then eventually it could be ported back to 2.1.

mpaperno, 2016-12-04:
I took a look and don't see anything preventing a quick port of this to 2.2, the relevant parts in 'next' are mostly unchanged.  I'll need to get 2.2 build set up to proceed (I hate CMake), so I'd like to know if this is worth pursuing.   I'd like to see it ported back to 2.0... eventually. :)

lshems, 2016-12-04:
@mpaperno:

if I understand it correctly, the compile will be done on ANY file loaded from the lua interpreter? so also a file that is loaded with the loadfile command?

Also, when using the string.dump(function), the code returned will be without the debug info, because the jit compiler is used?

Asking this to define how to get referenced files compiled.

As told, I'm not an expert on this at all, just an extensive user, and loading and unloading files to save memory still remains an important strategie to build complexer scripts.

Any info would be welcome.

mpaperno, 2016-12-04:
No, loadfile() is using the Lua base library directly, essentially bypassing what we've been doing here with the compiler.  We're actually saving the bytecode after (essentially) calling Lua's loadfile().

You could of course pass the .luac file name to loadfile() and it would load that.  But if you want to automate the compiling process from within your script, you'd need some kind of API exposed.  Like a "loadfile()" that does the same "JIT" compilation but with arbitrary scripts.  Wouldn't be hard to add at this point.

-Max

lshems, 2016-12-04:
Well, it would seem logical to have the same compiler behaviour throughout the LUA implementation. Since loading files from special created 'libraries' has already been used by several scripts, it would be a pity if loadfile() would follow the same logic.

I hope adding my viewpoint without being able to add to the code myself is not regarded as being lazy. Lol.

closed 2017-05-30 by kilrah