opentx/opentx #3319: Hungarian Sound Pack V1.2

hori64, 2016-02-26:
The Hungarian Sound Pack is missing from the V1.2.
We have prepared one of the Hungarian RC Forum:
https://drive.google.com/file/d/0B5pTbINRksqSNEtLZ3hpckRHYnM/view

But this package can only work perfectly when it is in the SOUNDS\en directory.
(I tried to create one based on the opentx-2.1.7\radio\src\translations\tts_hu.cpp, but it did not work properly)

I think the solution would be to tts_hu.cpp the same work as the tts_en.cpp.

Please, somebody help us!

projectkk2glider, 2016-02-26:
Are you sure that the grammar for numbers is the same in Hungarian and English. That is the purpose of separate tts files.

projectkk2glider, 2016-02-26:
I've just checked both tts files. They are basically the same, grammar is the same. Only the HU file has different constants which could be corrected to work with the new sound pack. 

But this will then brake the old Hungarian sound packs. Do you think this will be a problem?

projectkk2glider, 2016-02-26:
I see that no official 2.1 sound pack was made for Hungarian. So I think there is no problem changing this. Agreed?

hori64, 2016-02-26:
Yes. We want to use this pack.

hori64, 2016-03-02:
I built a development environment and modified the tts_hu.cpp.
I checked in the simulator. It works perfectly.
I'm not sure how well I uploaded this on Github.
I did it well?

hori64, 2016-03-02:
I do not have permission to add a new branch.
Here is the new tts_hu.cpp:
https://drive.google.com/file/d/0B5pTbINRksqSTVpOTEpEWWhyUGM/view?usp=sharing

kilrah, 2016-03-03:
Thanks, committed and pack uploaded!

closed 2016-03-03 by kilrah