opentx/opentx #3331: playFile("somesound.wav") sometimes skips playing on simulator

lshems, 2016-02-29:
It looks like the on some occasions code is advancing before the actual file is loaded and played in the simulator.

It is completely at random, but tends to be more frequent when executed directly after model simulation is started.

Difficult to reproduce, but on my system rather consistently happenes every now and then when playing around with the training scripts I build.

projectkk2glider, 2016-02-29:
* Do you see the playFile output in debug window?
* Does only a part of the file play or is the whole file is missing?

lshems, 2016-02-29:
First is difficult to see. I'll check
Whole file is not played.

projectkk2glider, 2016-02-29:
First thing is to check the debug output, it will also tell you about the  #3332 problem.

lshems, 2016-02-29:
I recently noticed the same behaviour for "loadFile", but there I can use a workaround if the file is skipped. Just reload until the variable to which the code is loaded is not empty anymore.

For playFile I cannot know if the file is played or not, so I have no workaround.

projectkk2glider, 2016-03-05:
Are you aware of the radio wide audio silence period of 0.5 seconds which takes effect:
 * when the radio is started
 * when `Reset Flight` is issued

This might be producing the result you are seeing. When anything is played during that time it will be silenced!

lshems, 2016-03-06:
Nop. It's not that. Thanks for thinking along.

I notice it when low on battery on my laptop, so it might have something to do with timings on file access or processor priorities in power saving mode.

I'll keep an eye open and post if I find some pattern.

There is sometimes a blocking of running of scripts due to a to many open files. This happened if having companion open for a long period of testing scripts. Only solution is to restart companion. Happens on 2.0 and 2.1. Posts as separate ticket?

projectkk2glider, 2016-03-06:
You can use Sysinternals tools (*handle* seems appropriate https://technet.microsoft.com/en-us/sysinternals/handle) to see how many files are open by the Companion process.

projectkk2glider, 2016-03-18:
There are a couple of thins missing from `StopMainThread()` which are leaving open files:
 * Lua is not deinitialized
 * currently playing audio file will stay open