opentx/opentx #3332: functionDoThis = load("somescript.lua") is not loading scripts always

lshems, 2016-02-29:
when loading code in scripts in LUA in the simulator, sometimes the load is not executed. 

it is completely at random, and seems to be more frequent when code is loaded directly after start of simulation.

difficult to debug, but probably related to the occasional skip of playing sound files in lua on the simulator.

projectkk2glider, 2016-02-29:
>when loading code in scripts in LUA

Please elaborate. What exactly are you doing.

lshems, 2016-02-29:
when executing 

`functionDoThis = load("somescript.lua") `

functionDoThis is sometimes empty, and has not loaded "somescript.lua", even when trying to load the same somescript.lua at exactly the same way as the try before.

Therefore, I use the following workaround:

```
taskBase.loadfile = function(file) --bugfixing!!!!!
  local loaded
  loaded=loadfile(file)
  while not loaded do
    loaded=loadfile(file)
  end
  return loaded
end
```

This will simply repeat the load until it is sure that the code is loaded.

I recently noticed the same behaviour for "playFile", but there I cannot use a workaround if the file is skipped being played.

It is now consistent on my uncompiled toolset. So no loop, no function load.

It seems that the load is not finished before the interpreter is going on executing.

There is no "file not found" error,so the code is executed as IF the file was loaded.

projectkk2glider, 2016-03-04:
First, function `load()` is not meant to be used to load Lua source from file.

The function `loadfile()` is appropriate and this function will return two values in case of error:
 * nil
 * error message

Please use `error message` in your script and report back what it is:
```lua
  local loaded
  local err
  loaded, err = loadfile("foo.lua")
  if loaded == nil then
    print ("err: " .. err)
  else
```

lshems, 2016-03-05:
sorry for the confusion, offcourse I meant loadfile(), as you probably noticed in the snippet.

I've added the error message, as for now I cant reproduce. It happens every now and then.

Ill report as soon as it starts giving error messages :)

Thanks for the help so far in trying to get behind the issue

lshems, 2016-11-29:
Ok, I think i figured out the cause: Low battery on a W10 laptop causing battery saving mode causing lag in file read response. So probably linked to W10.

projectkk2glider, 2016-11-30:
Let's close it then, Reopen if the problem reoccurs.

closed 2016-11-30 by projectkk2glider