opentx/opentx #3333: 9XR models/ settings not read by Companion 2.0

obguy99, 2016-03-01:
1)No problem writing models to radio but can't read model settings to companion.
See below. m128 running with 2.019 with EEPR 217.
Is this radio not supported? (see 2nd image)

If I write a model on companion and save it onto my computer I can not later open that file (to modify) and then rewrite to radio. (see first image) below
[image]

Started AVRDUDE
C:/Program Files (x86)/OpenTX/Companion 2.0/avrdude.exe -c usbasp -p m128 -P usb -U eeprom:r:C:/Users/POPROX~1/AppData/Local/Temp/4272-temp.hex:i
=================================================================================

avrdude.exe: error: programm enable: target doesn't answer. 1 
avrdude.exe: initialization failed, rc=-1
             Double check connections and try again, or use -F to override
             this check.

kilrah, 2016-03-01:
Do you have the correct radio type selected? How are you saving that model?

obguy99, 2016-03-01:
I download the firmware for 9XR m128 (it is a stock unmodified radio). I'm almost sure that is correct. I 1)write a model into the companion then 2) click save as (name on my desktop) then 3) click write models/settings to radio. That part works.
What I can't do is 1)Read models /settings from radio or 2) open the file that I had "saved as". Therefore each time I want to write a model I am starting from the beginning.
:) 
It's still better than the stock firmware!
Are there some specifics for the m128 stock 9XR that I am missing?

obguy99, 2016-03-01:
forgot to mention that I can only write firmware to radio if I uncheck (check firmware) and uncheck backup. Works well if I uncheck those.

[image]

kilrah, 2016-03-01:
Looks like it might be a problem with your programmer. Please continue this discussion on forums, you'll likely get better help there and this issue tracker is meant for OpenTX development discussions, not user support.

obguy99, 2016-03-02:
Understood, thanks! What about the issue of writing the model on companion and saving it to my computer and then later trying to open it? Shouldn't companion be able to open that file, regardless of the programmer or even the radio? I'm going to try an Arduino uno programmer. Thank you!

kilrah, 2016-03-02:
Ah yes, that is an issue indeed.

obguy99, 2016-03-02:
Ok. So 2.1 differently saves the models in "save as" and then opens them later without difficulty? That would help a ton since for my simple needs I can write to but not read with the companion to 9XR. At least I can then modify previous work and not start over each time.

obguy99, 2016-03-02:
Confirmed - saving to computer and can reopen that file for future editing works in 2.1.
 :) 
Happy guy since that creates a long term work flow with 9xr

closed 2016-03-24 by kilrah