opentx/opentx #3334: add field unit to LUA getFieldInfo

lshems, 2016-03-01:
Currently there is no way to know in the LUA environment which unit is assigned to a telemetry field. This causes problems when writing scripts for for instance launch altitude announcement.

In 2.0 there was automatic conversion on playValue for instance, which sort of "solved" this problem.

My suggestion would be to add the telemetry field unit, as specified in the telemetry settings, to the getFieldInfo() command.

lshems, 2016-03-01:
To be more specific:

in 2.1, sensor discovery will assign the units according to the general.imperial setting.
getValue(sensorfield) will return the corresponding value, so meters if set up in meters, and feet if set up in feet.

There is however no way to detect if the user has changed the sensor units after discovery.

So when determining what to play using playValue, or what to show using lcd.drawNumber, one can only guess based on the imperial setting what the sensor units probably are, and act accordingly.

This is actually worse then in 2.0, where at least the conversion was not done at sensor value level, so one could use the imperial setting to know how and if to convert for lcd.drawnumber, and playvalue had automatic conversion. 

by adding the unit number to the getFieldInfo, this could be solved.

lshems, 2016-03-01:
this would allow to use playValue() as in the special function, where the value unit is added in the announcement.

nigelsheffield, 2016-03-02:
I think it would make sense that changing the tx to imperial or metric would then change the units too and not only if you rediscover sensors.
Maybe there is some reason to do it like it is now?

lshems, 2016-03-02:
@nigel

I would prefer having a warning pop-up if changing to or from imperial settings, asking if you want to rediscover sensors.

There may be good reasons to have settings differ per sensor from the global setting.

kilrah, 2016-03-02:
The global setting is now only used to choose what unit is selected by default when a new sensor is discovered. They can be changed later by the user or even have the same value displayed in more than one unit, so the global setting can't be used anymore.

lshems, 2016-03-02:
Please leave as default sensor units.

kilrah, 2016-03-02:
I was talking of removing it from the lua interface, not from the radio. It's now useless to be able to read that from lua. 

> It would allow to choose your settings and just rediscover sensors to be in sync again.

No it doesn't work like that, you'd have to delete all of them and anything that's set up around them before rediscovering.

lshems, 2016-03-03:
Please leave it in the LUA interface. It is rather usefull to knwo what the users preferences are.

And without a LUA access to the sensor units, this is the best guess you get at knowing what the units are you are getting when doing a getvalue.

lshems, 2020-04-08:
Close, but not because stale??

Add unit to lua getFieldInfo #4750

closed 2020-04-08 by kilrah