opentx/opentx #3335: No Mavlink option on 9XR-Pro for 2.0.19

pyr8, 2016-03-01:
I upgraded to the 9XR-Pro from the 9XR. There is no option for "Mavlink" under "OpenTX for 9XR-PRO" in Companion 2.0.19.

Do I need to flash the 9XR-PRO with the "OpenTX for 9XR with M128 chip" in order to get Mavlink back?

kilrah, 2016-03-02:
Mavlink protocol support has not been developed for ARM-based radios. 
You can't flash a Pro with the firmware for a 9XR, completely different processor and architecture. 

Closing as duplicate of #2354.

closed 2016-03-02 by kilrah

pyr8, 2016-03-02:
It has been done apparently as per: https://plus.google.com/109713849321222194668/post/MYdHi19m1jh

kilrah, 2016-03-02:
Broken link.

pyr8, 2016-03-02:
https://goo.gl/Dd54sT

pyr8, 2016-03-02:
I'm using APM with Openlrs TX & RX and would like to have MAVLINK data displayed on the Pro. Is it at all possible, or am I SOL using the current hardware?

kilrah, 2016-03-02:
As far as we know, as explained there is currently no solution to have native mavlink on a 9XR-Pro. Your hardware is fine, but the software work necessary to make it work has not been done, the issue I linked is about other requests for it. 

Your link however suggests someone has made the necessary work but without publising it... it could be integrated into OpenTX if someone could chase him down, but it seems several people have asked what he's using and he doesn't reply.

geofrancis, 2016-03-17:
I am using native mavlink and FRsky D mode with my 9xr and 9xr pro, the firmware is still a work in progress.

https://github.com/adriano65/opentx-2.0.14

pyr8, 2016-03-21:
geofrancis, I have a 9XR-Pro and both TX/RX modules are orange openLRS from HK flashed with gitsly fork. How do I get MAVLINK data showing on the pro?