opentx/opentx #3340: Watchdog reset on Horus and settings restore

projectkk2glider, 2016-03-04:
This issue is about the Watchdog reset mechanism on Hours and for algorithms that enable persistence of radio and current model settings when WDT reset happens.

Current idea is (as discussed on gitter):
 * after every setting change:
  * write settings to SD card (already implemented)
  * take care not to leave SD card based settings in a corrupted state if a WDT reset happens in the middle of file write
  * store a subset of radio settings and currently active model settings (only relevant parts) in battery backed SDRAM (4kB available). The current calculation shows that all settings can not be saved, so only important setting will be stored here. Also settings will need to be compressed to fit.

 * after WDT reset:
  * try to restore settings from battery backed SRAM (validity verified by checksum).
  * if settings in SRAM are bad: initialize SD card an load only radio/model settings from SD card. 
  * all other SD card based functions are disabled (bitmaps, sounds, logs, etc...)

There is also another issue to consider. When battery backup SRAM is enabled, the battey consumption goes up (when radio is powered off). http://stackoverflow.com/questions/20667754/how-to-use-backup-sram-as-eeprom-in-stm32f4 Therefore we need to:
 * enable BB SRAM on every startup
 * disable BB SRAM on every power down

kilrah, 2016-03-04:
> When battery backup SRAM is enabled, the battey consumption goes up (when radio is powered off)

...but keeping all things in perspective - it would "only" halve the battery life that currently sits at 4.5 years typically.

projectkk2glider, 2016-03-11:
Tested on pre-production radio with commit 03988a3a623d0ac5f62beaf6d1b4544e2af06c17. Works.

kilrah, 2016-03-31:
Got stuck in a reboot loop
I forgot about the PXX on external not being implemented, selected it, and bam WDT, enter emergency mode, and... re-WDT forever so somehow the backup RAM was written during those 1.5sec between menu selection and wdt reset.
Need to check the write timer, it might not always be reset on change

kilrah, 2016-05-02:
Done, but leaving it open due to the potential bad read of stick mode that was reported

projectkk2glider, 2016-05-02:
This should fix stick mode problem #3455

projectkk2glider, 2016-05-09:
Stick mode problem was resolved.

closed 2016-05-09 by projectkk2glider