opentx/opentx #3344: Add lua function getLanguage() to be able to make script texts match transmitter language.

lshems, 2016-03-09:
There are two places for language definition:

In compile options for user interface. Covered in earlier request to make used compile options accessable from lua.

In language for sound files.

Please make those available for proper localisation of scripts.

lshems, 2016-03-09:
Language,voiceLanguage=getLanguage()
?

lshems, 2016-03-09:
And 
language=getlanguage("text")
VoiceLanguage=getLanguage("voice")

bsongis, 2016-03-11:
Will be only present in 2.2.0

projectkk2glider, 2016-03-11:
My mistake, I did not see the commit and thought that issue was closed by mistake.

bsongis, 2016-03-11:
It's completely untested

projectkk2glider, 2016-03-11:
I will reopen and close it after I test it.

projectkk2glider, 2016-03-13:
I do not like this implementation for two reasons:
* it defines two new functions, but the values could be returned in existing function `getGeneralSettings()`
* the `getTTSLanguage()` is too complicated and does not work, the `g_eeGeneral.ttsLanguage` is not set until user select the language for the first time. I would be better to use `currentLanguagePack->id` directly.

I will prepare a pull...

lshems, 2016-03-13:
Wow, nice solution!

projectkk2glider, 2016-03-13:
Thanks, my goal is to not introduce new Lua functions unless absolutely necessary.

lshems, 2016-03-13:
Well, mine is to be able to write scripts using the transmitters GUI settings for voice and text. Now I can.

This solves the other issue as well. Close it down?

And there is one for getting the compile parameter known in LUA, Sort of same reason. Cant see of another compile option useful to know? NoGVARS and NoOverridechannel perhaps can be tricky, as well as noheli. MMM, better leave that one open.

closed 2016-03-13 by projectkk2glider

lshems, 2016-03-13:
Update Milestone 2.1.8 ??

projectkk2glider, 2016-03-13:
No, this will be in 2.2.0

lshems, 2016-03-13:
Pity, but thanks anyway. Looking forward ;)

commit 086204ae (2016-03-11): Fixes #3344
PR #3358 (2016-03-13): Fixes #3344: radio and voice language returned in Lua getGeneralSetti…
Closes #3344 

Test script:
```lua
local function run(event)
  local s = getGeneralSettings()
  print("lang: "..s["language"])
  print("tts: "..s["voice"])
  return 1
end

return { run=run }
```
Result:
```
lang: EN
tts: sk
Script finished with status 1
```
commit 7ec2c2c5 (2016-03-13): Merge pull request #3358 from opentx/projectkk2glider/issue_3344_lua_language

Fixes #3344: radio and voice language returned in Lua getGeneralSetti…
commit d5db9b74 (2016-03-13): Fixes #3344: radio and voice language returned in Lua getGeneralSettings()