opentx/opentx #3351: Companion Sync SD card no syncing

pointu, 2016-03-11:
Hi,
Using SD card sync, but there is no syncing from a changed file on pc to old one one SD card (old old, lot of days).

W7 / OTX 2.1.7 / Companion 2.1.7

projectkk2glider, 2016-03-13:
Please provide more data, like the dates for affected file both on SD card and on computer.

pointu, 2016-03-13:
[image]
File test1.lua not synced
The opposite don't work also
[image]

But it work in both sens when file is missing on the other side.

projectkk2glider, 2016-03-13:
Thanks for additional info. This one is strange...

projectkk2glider, 2016-03-13:
@pointu if the file contents are the same, the files will not be copied, even if their date differs. Please confirm that your `test1.lua` files have different contents.

pointu, 2016-03-13:
I confirm there is a different contents, i add a comment

projectkk2glider, 2016-03-13:
I can't find any reason (in the source code) why it is not working for you. Can you provide additional information like: 
 * does the name of the files matter
 * does the position of file matter 
 * does only happen for particular dates
 * try the same thing with English language selected
 * are more files affected or just this one.

pointu, 2016-03-13:
i change the language for english > nothing more
I change the system date/time to utc > nothing more
I update a lot of files  > nothing more
i try for a long time, it never work  > nothing more
all files are affected and in other directory too

pointu, 2016-03-14:
Verifying with syntoy :
[image]
With Companion:
[image]

pointu, 2016-04-14:
No more to say
Test again and again in different ways, nothing more.
Is there a way to obtain the list of files with there attributes Companion get to go forward ?
(i'm not a dev so it's impossible for me to modify obtain and use the code)

W7 / OTX 2.1.8 / Companion 2.1.8

pointu, 2016-04-15:
Does someone can give me a test version that give me a log of sync to analyze like files in both sides like companion see them ?

projectkk2glider, 2016-04-15:
I will prepare something for you, hold on...

pointu, 2016-04-15:
Thanks for all
The goal is to go forward or close !

projectkk2glider, 2016-04-15:
While implementing a debug logging of sync process, I discovered a bug. It is now fixed. @pointu please check with the next nightly build (which I will start now).

pointu, 2016-04-15:
Will test that

pointu, 2016-04-15:
N771 or N711  ?

projectkk2glider, 2016-04-15:
Ups, 711 which is now ready.

pointu, 2016-04-15:
Think u do a loop, look > 
[image]
U write first in the correct way, but u write again in opposite way

That's why u have to test if the size have change, because the date/time have are few second different but the size are equal and i think it's not a good idea because u can have same weight for a file with different contents.

pointu, 2016-04-15:
More
It arrived only with the first file when there is more than one file not with the  other

pointu, 2016-04-15:
The result is 
[image]
Because the date/time are not equal

projectkk2glider, 2016-04-15:
I see another issue. Reopening

projectkk2glider, 2016-04-15:
Please try with N712. It will be ready in about 15 minutes.

pointu, 2016-04-15:
Better,
But the last... The modified date of the copied file is the date of the copy, not the date of the original file file, so they're always a different between the source and the target and that's not correct to verify who's the last
[image]
This is the result of sd sync, the file is the same but the last modified timestamp is not good.

projectkk2glider, 2016-04-16:
>The modified date of the copied file is the date of the copy, not the date of the original file file

This is how the current algorithm is supposed to work. Not sure it is the best/correct way. I will try setting the modified date.

kilrah, 2016-04-16:
The date of the copy should be the date of the original, otherwise every copied file will always be detected as modified on the next sync.

projectkk2glider, 2016-04-16:
Perhaps the current algorithm to detect change is flawed. It first checks for the modified date difference and proceeds only if they are different. So now we can have two files with the same name and modified date, but different contents and they won't be synced.

projectkk2glider, 2016-04-16:
One must also be careful, not all file-systems have the same modified date/time resolution.

pointu, 2016-04-16:
How could you have "two files with the same name and modified date" with different contents ? the last date/time is the good date/time thats how sync programs like synctoy work, otherwise you have to do a contents check

pointu, 2016-04-16:
what did i do close ?

elecpower, 2016-04-16:
> One must also be careful, not all file-systems have the same modified date/time resolution
Not ideal but assuming there is no way to solve and @kilrah point on flip-flop syncing what about a small time tolerance eg 1 sec?
The files being synced are highly ulikely to be changing within the date/time resolution variance on both the source and destination and remain the same size.

projectkk2glider, 2016-04-29:
I think the current state of the sync algorithm is OK. If two files have different date, but have the same contents, their dates won't be updated. This is how @bsongis originally wrote the algorithm.

commit 819e41c7 (2016-04-15): Fixes #3351: same file was used as source and destination in file contents comparison
commit e33e83f2 (2016-04-15): Fixes #3351: same file was used as source and destination in file contents comparison
commit e415be95 (2016-04-15): Fixes #3351: QByteArray used instead of QString for file contents comparison
closed 2016-04-29 by projectkk2glider