opentx/opentx #3354: OpenTx 2.1 option to move sensors

flarssen, 2016-03-12:
As sensors seems to be referenced by index, it would be helpful to be able to change the (random) order after discovery. Then telemetry settings can be reused when setting up new models,

projectkk2glider, 2016-03-13:
This looks like a dangerous function - ie an unsuspecting user (after setting up logical switches that use telemetry values) using this, moving a sensor would *corrupt* its logical switches, perhaps even without knowing.

kilrah, 2016-03-13:
Well it's implied that everything using those would be automatically shifted... but we already said we wouldn't venture into that dangerous territory.

flarssen, 2016-03-14:
Tools for corruption are already there. In fact they can be used to move sensors. Just more button pushing compared to a "move", like in the models list. 
"Copy" will create a copy in the first empty slot, so copy sensor3, delete sensor3, copy sensor5 and delete sensor5 will move sensor5 to position 3 and sensor3 to the bottom of the list (or first empty slot).

@kilrah 
Automatic shifting?

kilrah, 2016-03-15:
There is no "direct" relation between a sensor and its uses so automatic shifting would require going through hundreds of places in the radio that use a source to find if one is used and change it. 

The logic would be complex and a maintenance hell so we said we wouldn't do that.

flarssen, 2016-03-15:
I see, but that was not part of my suggestion. On the contrary, the idea is to adapt the list order to existing references (LS's and telemetry screens) in cases where they are not in sync, and also make it possible to have similar sensor order on all models, which makes it easier to find sensors when scrolling through the list of maybe 10-15 sensors with their min and max variants.

closed 2020-07-08 by stale[bot]