opentx/opentx #3356: Can't write current file to Taranis with OpenTX Companion 2.1.7

tinue, 2016-03-13:
It appears that whenever I click "Write Models and Settings To Radio", an invisible default file containing a "Model01" and default hardware settings is written to the Taranis, instead of the currently opened file. These are my steps to reproduce:

- First, I do a factory reset, and calibrate the sticks.
- Then I put the Taranis in bootloader mode (firmware & loader are v 2.1.7)
- I connect the Taranis to the PC, and do a "Read Models and Settings From Radio"
- I open the "General Settings" page, and change just one thing: Stick mode to "Mode 2"
- I click "Store calib. and hw settings in selected profile", and close the window
- I save my profile, and then click "Write Models and Settings to Radio". I check "Patch HW settings from profile"
- Progress goes to 100%, I disconnect and restart the Taranis

The Taranis has a default hw profile now, which has nothing to do with the profile that I created.
When I reconnect the Taranis and load its settings, I can see that e.g. the battery levels are reset to 9-12 Volts, from the 6-8v that were in the profile before.

I actually also can't write any models: All I get is the default "Model01", even if my file contains four models.

This happens on both a PC and a Mac, with OpenTX Companion 2.1.7.

projectkk2glider, 2016-03-13:
I assume the problem only appears if you use `Patch HW settings from profile` option. Right?

tinue, 2016-03-13:
No, even if I leave the hardware alone, and only try to update a model, I see the same behavior. 
If I load my initial model into a new file, and e.g. just rename it and write it back, I see no change on the Taranis.
Even there it looks like a default, empty file is written back.

projectkk2glider, 2016-03-13:
Then something basic is wrong, most likely you chosen wrong radio in the profile settings. Make sure you have selected Taranis as the radio!

tinue, 2016-03-13:
Sorry, my bad ;) 
You were right: It only happens if "Patch HW settings from profile" is checked. 

As far as I can tell, I have no errors in my profile.