opentx/opentx #3360: open tx firmware upgrade issue PLEASE HELP ...

facehead, 2016-03-14:
Hello there , 

really sorry about this post im sure iv made some stupid error as i was obviously excited to get into the new radio asap...

I recently bought a U.S.A radio that had been flashed with a German language. (im based in Germany)

everything was working fine but the menus were all in german etc..

serial as follows 

X9E001120A01
C02 HKHK
9/10/2015

I quickly looked online for language changing setting and instantly found i had to flash the radio with the new firmware to change the menu text to english 

so i tried 1st on my mac with the 2.1.7 companion and firmware from this page
http://www.open-tx.org/2016/01/04/opentx-2.1.7

with the horizontal trim to centre and power on method.

the firmware seemed to flash correctly and i rebooted the radio.

then i got vibration motors at full , and voice throttle warning switch warning , with the lower display going threw a list of warnings also.

i then tried the power off method on both osx and windows//////

next i got the same issue only this time NO SCREEN DISPLAY , just voice warning and vibrating....

aghhhhhhhhhhhhh

and now the horizontal trim to centre and power on method does nothing ...... ......

please help and please dont tell me its bricked , really hoping someone here can help me out or lead me to where i can find help !!
many many thanks !!!

: eek:

closed 2016-03-14 by facehead

projectkk2glider, 2016-03-14:
You need to select proper radio (Taranis 9XE) in the profile settings before downloading and flashing firmware. And use forums for such questions.

facehead, 2016-03-14:
yep i worked this out about 1 min after this post .. 
sorry

thanks