opentx/opentx #3361: Date and time resets after off x9e

free6k, 2016-03-14:
Date and time resets always after off x9e

This happens after some time after switching off, if you just turn the time is not reset

I have taranis x9e EU 2015
OpenTX 2.1.7 non-eu version

I read about this problem fixed in 2.1.2 version?

Thanks for answers!

kilrah, 2016-03-14:
Some early X9Es were delivered with a dead clock battery, check that first.

free6k, 2016-03-14:
Thanks for you answer! Where i can see taranis x9e scheme for check battery? Do you have manuals or photo? )

hrenz, 2016-03-14:
Its under the Batterie
 think its a CD2032 RTC Batterie

free6k, 2016-03-14:
@hrenz thanks you!! Great answer! I will try tomorrow replace battery... I have CR2032 at home. CR2032 and CD2032 different batteries? If yes, i will give answer tomorrow.

kilrah, 2016-03-14:
AFAIK it's a 1220, not 2032.

projectkk2glider, 2016-03-15:
Also check that you have a capacitor (circled in below photo) installed, this one is vital to the clock preservation.
[image]

hrenz, 2016-03-15:
Yes,  of course .
I have  changed this condensator too, becaue RTC Batterie was ok
470µF/16V or so

the original Cap was to short

free6k, 2016-03-15:
I today buy battery cr1220. I will change tomorrow. the capacitor is available. Thanks for good support!

free6k, 2016-03-16:
Thank you all for answers. I changed battery. In during all day i checked the time in x9e. Work very well! The problem can be closed

closed 2016-03-16 by projectkk2glider